openthread/ot-nrf528xx #41: Too many options are turned on by default

simonlingoogle, 2021-05-06:
Many OT options are `ON` by default, especially for `nRF52833` and `nRF52840`, e.g.

https://github.com/openthread/ot-nrf528xx/blob/993dc42f87cb905014593f417384eca33d2c6e6c/script/build#L52-L66

It's tedious that we have to turn off these options one by one for builds that don't need these options.

**Should we remove all these options because users can still specify them when they need it?**

I think this repository and the `script/build` tend to be general for all nRF528xx builds. 

Thoughts? @LuDuda @jwhui

bukepo, 2021-05-06:
Sounds good to me. Just make sure we keep the old options in the check-size script.

simonlingoogle, 2021-05-20:
> Sounds good to me. Just make sure we keep the old options in the check-size script.

All options above, except `-DOT_IP6_FRAGM=ON`, have been explicitly enabled for `check-size`.

Seems not a big issue to lose `-DOT_IP6_FRAGM=ON`. @bukepo

LuDuda, 2021-05-20:
Looks good to me, I already accepted PR, thanks! This also aligns better with the old behaviour.

PR #56 (2021-05-21): [build] remove default options
This commit removes default options. 

See #41 for more context. 

Fixes #41
closed 2021-05-21 by jwhui