openthread/ot-nrf528xx #69: Compilation error: conflicting types for 'otPlatRadioEnableCsl'

zwx1995esp, 2021-06-04:
I pull the new code and make the project using:
`./script/build nrf52840 USB_trans -DOT_CSL_RECEIVER=ON -DOT_LINK_METRICS=ON -DOT_DIAGNOSTIC=ON -DOT_THREAD_VERSION=1.2`
and I get a compilation error: conflicting types for 'otPlatRadioEnableCsl'

The cause of the problem seems to be the change in the commit `b0cb90b1`, the definition of function `otPlatRadioEnableCsl` has been changed. 

If I want to use the CSL feature, which parameters need to be added to the compile command?

jwhui, 2021-06-04:
@zwx1995esp , did you update the git submodule?
```
$ git submodule update
```

zwx1995esp, 2021-06-07:
Thank you, I am sorry to forget that I did not update the submodule, Now, it works.

closed 2021-06-07 by zwx1995esp