openthread/ot-nrf528xx #71: [radio] ot-daemon cannot start with nRF52840 RCP device.

wgtdkp, 2021-06-07:
Starting `ot-deamon` fails with `RadioSpinelNoResponse` when using the nRF52840 DK as a RCP device.

- `ot-daemon` commit: 87e6cf59913a9b247086a81f3d4c58205bc34725
- `ot-nrf528xx` commit: b00c50fed9b8353719ab1807f9d780fb595490b3

Steps to reproduce:
1. Build nRF52840 RCP (DK):
```shell
cd ~/ot-nrf528xx
./script/build nrf52840 UART_trans -DOT_SRP_CLIENT=1 -DOT_ECDSA=1
```
2. Build `ot-daemon`:
```shell
cd ~/openthread
cmake -S . -B build -GNinja -DOT_PLATFORM=posix -DOT_THREAD_VERSION=1.2 -DOT_COMMISSIONER=ON -DOT_BORDER_AGENT=ON -DOT_JOINER=ON -DOT_SRP_SERVER=ON -DOT_ECDSA=ON -DOT_SERVICE=ON -DOT_DAEMON=ON -DOT_BORDER_ROUTING=ON -DOT_BORDER_ROUTER=ON -DOT_DAEMON=ON -DOT_BORDER_AGENT=ON -DOT_COMMISSIONER=ON -DOT_REFERENCE_DEVICE=ON -DOT_FULL_LOGS=ON
```
3. Start `ot-daemon` and fail:
```shell
sudo ./build/src/posix/ot-daemon -d 6 -v -B wlan0 'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/src/posix/ot-daemon[25914]: Running OPENTHREAD/20200818-00989-g87e6cf599; POSIX; Jun  7 2021 11:46:00
./build/src/posix/ot-daemon[25914]: Thread version: 3
./build/src/posix/ot-daemon[25914]: [DEBG]-MAC-----: Idle mode: Radio sleeping
./build/src/posix/ot-daemon[25914]: [DEBG]-PLAT----: Wait response: tid=2 key=2048
./build/src/posix/ot-daemon[25914]: [CRIT]-PLAT----: HandleRcpTimeout() at ../src/lib/spinel/radio_spinel_impl.hpp:2190: RadioSpinelNoResponse
```

jwhui, 2021-06-07:
@wgtdkp , I was able to start `ot-daemon` successfully with the steps you provided.

Did you [disable Mass Storage Device](https://github.com/openthread/ot-nrf528xx/tree/main/src/nrf52840#mass-storage-device-known-issue)?

wgtdkp, 2021-06-08:
> @wgtdkp , I was able to start `ot-daemon` successfully with the steps you provided.
> 
> Did you [disable Mass Storage Device](https://github.com/openthread/ot-nrf528xx/tree/main/src/nrf52840#mass-storage-device-known-issue)?

@jwhui Thanks! It is exactly the `Mass Storage Device` issue. The daemon works perfectly after disable it!

closed 2021-06-08 by wgtdkp