openthread/ot-nrf528xx #111: [mbedtls] How to rebuild the mbedTLS libraries

AuSchell, 2021-07-20:
I would like to rebuild the mbedTLS libraries in path thrid_party/NordicSemiconductor/libraries/nrf_security/lib, but i don't have any clue on how to to that, can somebody please help me?

To be more specific about what i would like to do: i want to use the "mbedtls_x509write_csr" methods, but those libraries seem to have been built without MBEDTLS_X509_CSR_WRITE_C or MBEDTLS_X509_CSR_PARSE_C defined, which is necessary enable them. Is there maybe another way i could achive this?

MarekPorwisz, 2021-07-21:
Hi,

The easiest thing would be to switch to using NCS. OpenThread is integrated there and you should be able to configure nrf_security. The disadvantage is that you would need to adapt your application to be able to work under Zephyr OS. Another disadvantage is that OpenThread fork in NCS/Zephyr is a little bit behind Vanilla OpenThread repo - we do the upmerges regularly, so it should not be far behind.

If you prefer to stay with current solution you still need to get NCS: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/gs_installing.html

Then you can follow our "draft" guidelines:

1. Checkout nrf to https://github.com/MarekPorwisz/fw-nrfconnect-nrf/tree/vanilla_ot_nrf_security
2. Call west update
3. Checkout nrfxlib to https://github.com/MarekPorwisz/sdk-nrfxlib/tree/vanilla_ot_nrf_security (always after west update)
4. In case of using python3.6 => sudo apt-get install libpython3.6-dev
5. python3.6 -m pip install numpy west pyelftools
6. cd nrf/samples/openthread/cli
7. ./build_libs.sh
8. Built nrf_security is in openthread_mbedtls_out

**Notes:**
Keep in mind there are files not related to nrf_security in openthread that are placed in the nrf_security directory - do not remove them accidentally.
Oberon backend can be replaced with vanilla backend in the vanilla_nrf_sec.conf file if needed.

**Hint:**
When debugging FW build with the libraries (e.g. bare OpenThread) the debugger will not list the code as the source file paths will not match.
In such case one can check out the nrf_security branch used to build the libraries with git and then perform the path substitution in gdb.
Assuming that there are still libraries build by Marek in /home/map5/repo/ncs the command is as follows:
set substitute-path '/home/map5/repo/ncs' '<your_ncs_path>'_

This will generate nrf_security for you. My forks were created based on NCS master not long after 1.4 release. If you wish to do the upmerge keep in mind that fw-nrfconnect-nrf (currently sdk-nrf) and sdk-nrfxlib need to be aligned. west.yml file in sdk-nrf indicates what version of sdk-nrfxlib is needed.

If I remember correctly samples/openthread/cli/vanilla_nrf_sec.conf file needs to be modified in order to enable/disable any additional features.

AuSchell, 2021-07-21:
Thank you for your quick response, i will try that out and write back if it has helped.

jwhui, 2021-09-14:
Closing stale issue.

closed 2021-09-14 by jwhui