openthread/ot-nrf528xx #112: nRF52840DK flashed as RCP fails to connect via ot-daemon

nageshshamnur, 2021-07-21:
Infrastructure:
RPi4B: enabling the openthread service via RCP  using nRF52840DK board. nRF52840 is flashed with rcp image as explained in the link: https://openthread.io/codelabs/openthread-hardware#3. 
command used to build RCP for nRF52840DK : ```script/build nrf52840 USB_trans```

after flashing as mentioned in the link above when i connect the DK board to my RPi4B and try to run ot-daemon as explained in the same link above, then i am getting error as below:
```
$ sudo ./build/posix/src/posix/ot-daemon -v
'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[7104]: Running
OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21
2021 12:46:40
./build/posix/src/posix/ot-daemon[7104]: Thread version: 3
./build/posix/src/posix/ot-daemon[7104]: Thread interface: wpan1
./build/posix/src/posix/ot-daemon[7104]: RCP version:
OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 20
2021 20:52:57
./build/posix/src/posix/ot-daemon[7104]: [CRIT]-PLAT----:
RestoreProperties() at
../../src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
```
then i changed the command to build RCP for nRF52840 as : ```./script/build nrf52840 USB_trans -DOT_SRP_CLIENT=1 -DOT_ECDSA=1```  solved the issue. subsequent connect to RCP via ot-daemon is successful with log as below:
```
$ sudo ./build/posix/src/posix/ot-daemon -v
'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[7639]: Running
OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21
2021 12:46:40
./build/posix/src/posix/ot-daemon[7639]: Thread version: 3
./build/posix/src/posix/ot-daemon[7639]: Thread interface: wpan1
./build/posix/src/posix/ot-daemon[7639]: RCP version:
OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21
2021 12:59:21
```
So, my question is:
1) is the documentation incorrect about building RCP image for nRF52840 ```script/build nrf52840 USB_trans```
2) is the ot-nrf528xx revision no: 0b95a69ec0ceabaf82e9b07d9e82e326597f7e38 is broken so that it is not running  for ```script/build nrf52840 USB_trans``` and but working when built using ```./script/build nrf52840 USB_trans -DOT_SRP_CLIENT=1 -DOT_ECDSA=1```

wgtdkp, 2021-07-21:
`-DOT_SRP_CLIENT=1 -DOT_ECDSA=1` are optional, I don't think it is required. can you provide the full log for more visibility?

nageshshamnur, 2021-07-21:
@wgtdkp : RCP build without ```-DOT_SRP_CLIENT=1 -DOT_ECDS=1``` is successful without any issues, issues happens when i connect to the board via USB and try to connect via ot-daemon log for which i have attached as above.

nageshshamnur, 2021-07-21:
apologies for the wrong description, i am facing problem when i build with Thread v1.2 as below:
```script/build nrf52840 USB_trans -DOT_THREAD_VERSION=1.2```

and i am getting error as below:
```
$ sudo ./build/posix/src/posix/ot-daemon -v
'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[7104]: Running
OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21
2021 12:46:40
./build/posix/src/posix/ot-daemon[7104]: Thread version: 3
./build/posix/src/posix/ot-daemon[7104]: Thread interface: wpan1
./build/posix/src/posix/ot-daemon[7104]: RCP version:
OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 20
2021 20:52:57
./build/posix/src/posix/ot-daemon[7104]: [CRIT]-PLAT----:
RestoreProperties() at
../../src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
```

i have updated the description in the original post as well.

nageshshamnur, 2021-07-21:
connection got established after multiple retires, hence closing the ticket.

closed 2021-07-21 by nageshshamnur