openthread/ot-nrf528xx #114: ot-daemon keep disconnecting when executing ot-ctl commands

nageshshamnur, 2021-07-21:
I have flashed RCP image on my nRF52840dongle as explained in the link: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/openthread_rcp_nrf_dongle.md
revision number for ot-nrf528xx is : 0b95a69ec0ceabaf82e9b07d9e82e326597f7e38

when i run the ot-daemon in a window and then try executing ot-ctl commands in another window, ot-daemon keeps exiting with errors as below:
```
rf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6555]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[6555]: Thread version: 3
./build/posix/src/posix/ot-daemon[6555]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6555]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[6555]: [CRIT]-PLAT----: Unexpected RCP reset: RESET_POWER_ON
./build/posix/src/posix/ot-daemon[6555]: [CRIT]-PLAT----: RestoreProperties() at /home/nrf-connect-sdk/CHIP/openthread/src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6614]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[6614]: Thread version: 3
./build/posix/src/posix/ot-daemon[6614]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6614]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[6614]: [CRIT]-PLAT----: Unexpected RCP reset: RESET_POWER_ON
./build/posix/src/posix/ot-daemon[6614]: [CRIT]-PLAT----: RestoreProperties() at /home/nrf-connect-sdk/CHIP/openthread/src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6630]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[6630]: Thread version: 3
./build/posix/src/posix/ot-daemon[6630]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6630]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[6630]: [CRIT]-PLAT----: otPlatRadioSetPanId() at /home/nrf-connect-sdk/CHIP/openthread/src/posix/platform/radio.cpp:188: Failure
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6679]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[6679]: Thread version: 3
./build/posix/src/posix/ot-daemon[6679]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6679]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[6679]: [CRIT]-PLAT----: RestoreProperties() at /home/nrf-connect-sdk/CHIP/openthread/src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6780]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[6780]: Thread version: 3
./build/posix/src/posix/ot-daemon[6780]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6780]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
ot-daemon: /home/nrf-connect-sdk/CHIP/openthread/src/core/mac/sub_mac.cpp:521: void ot::Mac::SubMac::HandleTransmitDone(ot::Mac::TxFrame&, ot::Mac::RxFrame*, ot::Error): Assertion `false' failed.
Aborted
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6824]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[6824]: Thread version: 3
./build/posix/src/posix/ot-daemon[6824]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6824]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[6824]: [CRIT]-PLAT----: Unexpected RCP reset: RESET_POWER_ON
./build/posix/src/posix/ot-daemon[6824]: [CRIT]-PLAT----: RestoreProperties() at /home/nrf-connect-sdk/CHIP/openthread/src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6866]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[6866]: Thread version: 3
./build/posix/src/posix/ot-daemon[6866]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6866]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[6866]: [CRIT]-PLAT----: RestoreProperties() at /home/nrf-connect-sdk/CHIP/openthread/src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6969]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[6969]: Thread version: 3
./build/posix/src/posix/ot-daemon[6969]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6969]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[6969]: [CRIT]-PLAT----: Unexpected RCP reset: RESET_POWER_ON
./build/posix/src/posix/ot-daemon[6969]: [CRIT]-PLAT----: RestoreProperties() at /home/nrf-connect-sdk/CHIP/openthread/src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[7101]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[7101]: Thread version: 3
./build/posix/src/posix/ot-daemon[7101]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[7101]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
ot-daemon: /home/nrf-connect-sdk/CHIP/openthread/src/core/mac/sub_mac.cpp:521: void ot::Mac::SubMac::HandleTransmitDone(ot::Mac::TxFrame&, ot::Mac::RxFrame*, ot::Error): Assertion `false' failed.
Aborted
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[7140]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[7140]: Thread version: 3
./build/posix/src/posix/ot-daemon[7140]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[7140]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
[... 57 lines cut ...]
./build/posix/src/posix/ot-daemon[7477]: Running OPENTHREAD/thread-reference-20200818-1088-g5e23211e0; POSIX; Jul 21 2021 17:34:07
./build/posix/src/posix/ot-daemon[7477]: Thread version: 3
./build/posix/src/posix/ot-daemon[7477]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[7477]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[7477]: [CRIT]-PLAT----: Unexpected RCP reset: RESET_POWER_ON
[... cut ...]
./build/posix/src/posix/ot-daemon[7514]: RCP version: OPENTHREAD/thread-reference-20200818-1079-g5f39ba055; NRF52840; Jul 21 2021 21:30:25
./build/posix/src/posix/ot-daemon[7514]: [CRIT]-PLAT----: Unexpected RCP reset: RESET_POWER_ON
./build/posix/src/posix/ot-daemon[7514]: [CRIT]-PLAT----: RestoreProperties() at /home/nrf-connect-sdk/CHIP/openthread/src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nrf-connect-sdk@root1-Nitro-AN515-55:~/CHIP/openthread$ 
```

commands which i am executing on ot-ctl are:
```
> scan
> dataset networkkey <key>
> dataset commit active
> ifconfig up
> thread start
```
kindly help

jwhui, 2021-07-22:
From the logs, it appears that the RCP is resetting unexpectedly.
```
./build/posix/src/posix/ot-daemon[6555]: [CRIT]-PLAT----: Unexpected RCP reset: RESET_POWER_ON
```

I tried to reproduce this issue using the latest main branch (098b6da38b040863d347f9a530e7930a686dc5d0), but was unable to reproduce the issue.

@nageshshamnur , is there anything that may cause your USB dongle to reset and/or power cycle?

@LuDuda @MarekPorwisz , thoughts?

nageshshamnur, 2021-07-23:
@jwhui : Thanks for the reply. i am not sure why reset power cycle has triggered? i am not doing anything extra other than executing ot-ctl commands in a different console. I have tried innumerable number of times. More than rest power event what i have seen more prominent is the following 2 issues (in both DK and dongle):

./build/posix/src/posix/ot-daemon[6555]: [CRIT]-PLAT----: RestoreProperties() at /home/nrf-connect-sdk/CHIP/openthread/src/lib/spinel/radio_spinel_impl.hpp:2296: Failure

ot-daemon: /home/nrf-connect-sdk/CHIP/openthread/src/core/mac/sub_mac.cpp:521: void ot::Mac::SubMac::HandleTransmitDone(ot::Mac::TxFrame&, ot::Mac::RxFrame*, ot::Error): Assertion `false' failed.

wgtdkp, 2021-07-23:
> ot-daemon: /home/nrf-connect-sdk/CHIP/openthread/src/core/mac/sub_mac.cpp:521: void ot::Mac::SubMac::HandleTransmitDone(ot::Mac::TxFrame&, ot::Mac::RxFrame*, ot::Error): Assertion `false' failed.

This seems to me that the daemon and RCP are incompatible with each other. Can use build the `ot-daemon` with the same revision of `ot-nrf528xx/openthread`?

jwhui, 2021-07-23:
> This seems to me that the daemon and RCP are incompatible with each other. Can use build the `ot-daemon` with the same revision of `ot-nrf528xx/openthread`?

@wgtdkp , the two commits are pretty close together:

ot-daemon:
```
commit 5e23211e048cf8fcc4038bcb4d3a876fb4044e21
Author: Simon Lin <simonlin@google.com>
Date:   Wed Jul 21 14:08:56 2021 +0800
```

RCP:
```
commit 5f39ba055bb5adcc28fd545226cbd6836a015c2e
Author: Jonathan Hui <jonhui@google.com>
Date:   Fri Jul 16 09:49:18 2021 -0700
```

I'm not aware of any incompatibility between those commits.

jwhui, 2021-07-23:
> @jwhui : Thanks for the reply. i am not sure why reset power cycle has triggered? i am not doing anything extra other than executing ot-ctl commands in a different console. I have tried innumerable [...]

@nageshshamnur , I noticed that you are building out of `nrf-connect-sdk`. Can you try cloning the OpenThread repos directly and building both `ot-daemon` and RCP straight out of the OpenThread repos?

nageshshamnur, 2021-07-23:
> @nageshshamnur , I noticed that you are building out of `nrf-connect-sdk`. Can you try cloning the OpenThread repos directly and building both `ot-daemon` and RCP straight out of the OpenThread repos?

i haven't used nrf-connect-sdk, i have done using the following repos:
https://github.com/openthread/openthread.git
https://github.com/openthread/ot-nrf528xx.git

nageshshamnur, 2021-07-23:
> @nageshshamnur , I noticed that you are building out of `nrf-connect-sdk`. Can you try cloning the OpenThread repos directly and building both `ot-daemon` and RCP straight out of the OpenThread repos?

that's just the linux account name, no relation to ncs whatsoever.

jwhui, 2021-07-23:
@nageshshamnur , unfortunately, I have not seen these issues. As mentioned above, I tested with the latest main branches for both repos and could not reproduce your issue. Can you try again with the latest main branches?

nageshshamnur, 2021-07-23:
> @nageshshamnur , unfortunately, I have not seen these issues. As mentioned above, I tested with the latest main branches for both repos and could not reproduce your issue. Can you try again with the latest main branches?

can you give me specific revision numbers?

jwhui, 2021-07-23:
> > @nageshshamnur , unfortunately, I have not seen these issues. As mentioned above, I tested with the latest main branches for both repos and could not reproduce your issue. Can you try again with the latest main branches?
> 
> can you give me specific revision numbers?

@nageshshamnur , the latest main branches are:
- https://github.com/openthread/openthread/commit/daf2ad7c245f947cadb2093af74ba9d121525473
- https://github.com/openthread/ot-nrf528xx/commit/098b6da38b040863d347f9a530e7930a686dc5d0

nageshshamnur, 2021-07-23:
> > > @nageshshamnur , unfortunately, I have not seen these issues. As mentioned above, I tested with the latest main branches for both repos and could not reproduce your issue. Can you try again with the latest main branches?
> > 
> > 
> > can you give me specific revision numbers?
> 
> @nageshshamnur , the latest main branches are:
> 
> * [openthread/openthread@daf2ad7](https://github.com/openthread/openthread/commit/daf2ad7c245f947cadb2093af74ba9d121525473)
> * [098b6da](https://github.com/openthread/ot-nrf528xx/commit/098b6da38b040863d347f9a530e7930a686dc5d0)

thanks for the quick reply. Will try a fresh build

nageshshamnur, 2021-07-23:
> @nageshshamnur , the latest main branches are:
>
> * [openthread/openthread@daf2ad7](https://github.com/openthread/openthread/commit/daf2ad7c245f947cadb2093af74ba9d121525473)
> * [...]

how about ot-br-posix ?

jwhui, 2021-07-23:
> > @nageshshamnur , the latest main branches are:
> > 
> > * [openthread/openthread@daf2ad7](https://github.com/openthread/openthread/commit/daf2ad7c245f947cadb2093af74ba9d121525473)
> > * [098b6da](https://github.com/openthread/ot-nrf528xx/commit/098b6da38b040863d347f9a530e7930a686dc5d0)
> 
> how about ot-br-posix ?

Latest ot-br-posix main branch is https://github.com/openthread/ot-br-posix/commit/2206d84aca281845288ef8b1e9b31f485f75ec77. Note that ot-br-posix does not make use of `ot-daemon`.

nageshshamnur, 2021-07-23:
I tried again with a fresh build as suggested:

instructions used to build openthread:
$ git clone --recursive https://github.com/openthread/openthread.git
$ cd openthread
$ ./script/bootstrap
$ script/cmake-build posix -DOT_DAEMON=ON

instructions used to build RCP for nRF52840DK:
$ git clone --recursive https://github.com/openthread/ot-nrf528xx.git
$ cd ot-nrf528xx
$ script/build nrf52840 USB_trans
$ cd ./build/bin
$ arm-none-eabi-objcopy -O ihex ot-rcp ot-rcp.hex
$ cd /opt/nrfjprog/
$ ./nrfjprog -f nrf52 -s <serial_dk> --chiperase --program \
       ~/src/ot-nrf528xx/build/bin/ot-rcp.hex --reset

log of ot-daemon:
```
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6524]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[6524]: Thread version: 3
./build/posix/src/posix/ot-daemon[6524]: Thread interface: wpan1
./build/posix/src/posix/ot-daemon[6524]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 11:52:19
./build/posix/src/posix/ot-daemon[6524]: [CRIT]-PLAT----: RestoreProperties() at ../../src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6547]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[6547]: Thread version: 3
./build/posix/src/posix/ot-daemon[6547]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[6547]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 11:52:19
ot-daemon: ../../src/core/mac/sub_mac.cpp:521: void ot::Mac::SubMac::HandleTransmitDone(ot::Mac::TxFrame&, ot::Mac::RxFrame*, ot::Error): Assertion `false' failed.
Aborted
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[6668]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[6668]: Thread version: 3
./build/posix/src/posix/ot-daemon[6668]: Thread interface: wpan1
./build/posix/src/posix/ot-daemon[6668]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 11:52:19
./build/posix/src/posix/ot-daemon[6668]: [CRIT]-PLAT----: RestoreProperties() at ../../src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
```

log of ot-ctl:
```
nagesh@Nagesh-RPi:~/CHIP/openthread/build/posix/src/posix$ sudo ot-ctl 
> dataset init new
Done
> dataset
Active Timestamp: 1
Channel: 16
Channel Mask: 0x07fff800
Ext PAN ID: 30b0d61660f723e0
Mesh Local Prefix: fd32:6ae9:edca:38bd::/64
Network Key: 533262b956cfab14edfb48bcc04e5ee3
Network Name: OpenThread-ec7a
PAN ID: 0xec7a
PSKc: 088e3b733e5ab48d1331ef99734e0e11
Security Policy: 672 onrcb
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
> 
```

nageshshamnur, 2021-07-23:
> @jwhui: I tried again with a fresh build as suggested, but issue still persists, i have added detailed steps which i used to recreate.
> 
> instructions used to build openthread:
> $ git clone --recursive https://github.com/openthread/openthread.git
> $ cd openthread
> $ ./script/bootstrap
> $ script/cmake-build posix -DOT_DAEMON=ON
> 
> instructions used to build RCP for nRF52840DK:
> $ git clone --recursive https://github.com/openthread/ot-nrf528xx.git
> $ cd ot-nrf528xx
> $ script/build nrf52840 USB_trans
> $ cd ./build/bin
> $ arm-none-eabi-objcopy -O ihex ot-rcp ot-rcp.hex
> $ cd /opt/nrfjprog/
> $ ./nrfjprog -f nrf52 -s <serial_dk> --chiperase --program
> ~/src/ot-nrf528xx/build/bin/ot-rcp.hex --reset
> 
> log of ot-daemon:
> 
> ```
> nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
> ./build/posix/src/posix/ot-daemon[6524]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
> ./build/posix/src/posix/ot-daemon[6524]: Thread version: 3
> ./build/posix/src/posix/ot-daemon[6524]: Thread interface: wpan1
> ./build/posix/src/posix/ot-daemon[6524]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 11:52:19
> ./build/posix/src/posix/ot-daemon[6524]: [CRIT]-PLAT----: RestoreProperties() at ../../src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
> nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
> ./build/posix/src/posix/ot-daemon[6547]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
> ./build/posix/src/posix/ot-daemon[6547]: Thread version: 3
> ./build/posix/src/posix/ot-daemon[6547]: Thread interface: wpan0
> ./build/posix/src/posix/ot-daemon[6547]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 11:52:19
> ot-daemon: ../../src/core/mac/sub_mac.cpp:521: void ot::Mac::SubMac::HandleTransmitDone(ot::Mac::TxFrame&, ot::Mac::RxFrame*, ot::Error): Assertion `false' failed.
> Aborted
> nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
> ./build/posix/src/posix/ot-daemon[6668]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
> ./build/posix/src/posix/ot-daemon[6668]: Thread version: 3
> ./build/posix/src/posix/ot-daemon[6668]: Thread interface: wpan1
> ./build/posix/src/posix/ot-daemon[6668]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 11:52:19
> ./build/posix/src/posix/ot-daemon[6668]: [CRIT]-PLAT----: RestoreProperties() at ../../src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
> ```
> 
> log of ot-ctl:
> 
> ```
> nagesh@Nagesh-RPi:~/CHIP/openthread/build/posix/src/posix$ sudo ot-ctl 
> > dataset init new
> Done
> > dataset
> Active Timestamp: 1
> Channel: 16
> Channel Mask: 0x07fff800
> Ext PAN ID: 30b0d61660f723e0
> Mesh Local Prefix: fd32:6ae9:edca:38bd::/64
> Network Key: 533262b956cfab14edfb48bcc04e5ee3
> Network Name: OpenThread-ec7a
> PAN ID: 0xec7a
> PSKc: 088e3b733e5ab48d1331ef99734e0e11
> Security Policy: 672 onrcb
> Done
> > dataset commit active
> Done
> > ifconfig up
> Done
> > thread start
> Done
> > 
> ```

while building RCP for nRF52840 DK, i have tried both the options, but issue still remains the same:
``` 
script/build nrf52840 USB_trans 
```
```
script/build nrf52840 USB_trans -DOT_THREAD_VERSION=1.2
```

LuDuda, 2021-07-23:
Hello @nageshshamnur,

When you try to build with  `USB_trans` option on nRF52840 DK, how do you connect the DK to the PC/RaspberryPi? Note that the DK has two USB connectors one for `Segger J-Link` (UART transport on nRF52840 MCU), or nRF USB (for which you are building the firmware).

[image]

Some more info: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_thread_tools.html#configuring-a-radio-co-processor

nageshshamnur, 2021-07-23:
> Hello @nageshshamnur,
>
> When you try to build with `USB_trans` option on nRF52840 DK, how do you connect the DK to the PC/RaspberryPi? Note that the DK has two USB connectors one for `Segger J-Link` [...]

Hi @LuDuda,
   I have build RCP image for DK on my ubuntu 21.04 laptop and then connect the same to my RPi. I have built using the instructions as explained in this page.
https://openthread.io/codelabs/openthread-hardware#3
Use segger j-link USB connector port while flashing the RCP image (with VDD enabled) and nRF USB connector (with USB mode) while running ot-daemon.

nageshshamnur, 2021-07-23:
For nRF52840 i tried UART transport as explained in the link: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_thread_tools.html#configuring-a-radio-co-processor and every time i execute the command ```dataset commit active``` on ot-ctl, ot-daemon exits and everytime at this line ```otPlatRadioSetMacKey() at ../../src/posix/platform/radio.cpp:573```

```
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[8722]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[8722]: Thread version: 3
./build/posix/src/posix/ot-daemon[8722]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[8722]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 15:09:07
./build/posix/src/posix/ot-daemon[8722]: [CRIT]-PLAT----: RestoreProperties() at ../../src/lib/spinel/radio_spinel_impl.hpp:2296: Failure
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[8764]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[8764]: Thread version: 3
./build/posix/src/posix/ot-daemon[8764]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[8764]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 15:09:07
./build/posix/src/posix/ot-daemon[8764]: [CRIT]-PLAT----: otPlatRadioSetMacKey() at ../../src/posix/platform/radio.cpp:573: Failure
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[8812]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[8812]: Thread version: 3
./build/posix/src/posix/ot-daemon[8812]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[8812]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 15:09:07
./build/posix/src/posix/ot-daemon[8812]: [CRIT]-PLAT----: otPlatRadioSetMacKey() at ../../src/posix/platform/radio.cpp:573: Failure
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[8916]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[8916]: Thread version: 3
./build/posix/src/posix/ot-daemon[8916]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[8916]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 15:09:07
./build/posix/src/posix/ot-daemon[8916]: [CRIT]-PLAT----: otPlatRadioSetMacKey() at ../../src/posix/platform/radio.cpp:573: Failure
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[8954]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[8954]: Thread version: 3
./build/posix/src/posix/ot-daemon[8954]: [CRIT]-PLAT----: otPlatSettingsGet() at ../../src/posix/platform/settings.cpp:288: Failure
nagesh@Nagesh-RPi:~/CHIP/openthread$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
./build/posix/src/posix/ot-daemon[8957]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:06
./build/posix/src/posix/ot-daemon[8957]: Thread version: 3
./build/posix/src/posix/ot-daemon[8957]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[8957]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Jul 23 2021 15:09:07
./build/posix/src/posix/ot-daemon[8957]: [CRIT]-PLAT----: otPlatRadioSetMacKey() at ../../src/posix/platform/radio.cpp:573: Failure
nagesh@Nagesh-RPi:~/CHIP/openthread$ 
```

so for me for nRF52840DK board, both USB and UART transport is not working

nageshshamnur, 2021-07-23:
@LuDuda : any pointers for me to check further upon? right now i am clueless

LuDuda, 2021-07-23:
@nageshshamnur I will check your steps on a desk and let you now.

LuDuda, 2021-07-26:
@nageshshamnur I was able to reproduce your issue only when i turned on MSD on the nRF52840 DK. Could you please connect the nRF52840 DK to your PC using only Segger J-Link USB and disable MSD (remember to power cycle the DK). MSD reduces the maximum size of the single USB transaction. Build the RCP image with `UART_trans`.

https://github.com/openthread/ot-nrf528xx/tree/main/src/nrf52840#mass-storage-device-known-issue

I used the same commands as you. Make sure that your settings didn't corrupt along the way by executing the following commands:
```
ot-nrf528xx/openthread$ sudo rm -rf tmp
ot-nrf528xx/openthread$ sudo ./build/posix/src/posix/ot-daemon -v 'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
```

I'm not able to reproduce the issue using USB transport.

nageshshamnur, 2021-08-02:
> @nageshshamnur I was able to reproduce your issue only when i turned on MSD on the nRF52840 DK. Could you please connect the nRF52840 DK to your PC using only Segger J-Link USB and disable MSD [...]

Hi @LuDuda,
    I tried the exact steps what you mentioned in the above post and even i tried in 2 different machines. Everytime, ot-daemon exits with the log as below:
```
$ sudo ./build/posix/src/posix/ot-daemon -v     'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200'
[sudo] password for nrf-connect-sdk: 
./build/posix/src/posix/ot-daemon[11299]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Jul 23 2021 11:43:00
./build/posix/src/posix/ot-daemon[11299]: Thread version: 3
./build/posix/src/posix/ot-daemon[11299]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[11299]: RCP version: OPENTHREAD/thread-reference-20200818-1089-gde53d721d; NRF52840; Aug  2 2021 17:41:52
./build/posix/src/posix/ot-daemon[11299]: [CRIT]-PLAT----: otPlatRadioSetMacKey() at /home/nrf-connect-sdk/CHIP/openthread/src/posix/platform/radio.cpp:573: Failure
```
it happens when i execute the command ```dataset commit active``` on ot-ctl. Series of commands as below:
```
$sudo ot-ctl
> dataset init new
Done
> dataset
Active Timestamp: 1
Channel: 11
Channel Mask: 0x07fff800
Ext PAN ID: 34360dbd434e9b8f
Mesh Local Prefix: fd64:171e:be8f:f702::/64
Network Key: 4b99c48b6710232256485b3e488ebde1
Network Name: OpenThread-f01c
PAN ID: 0xf01c
PSKc: 59ebced131e0651211eb02ed2d891fd2
Security Policy: 672 onrcb
Done
> dataset commit active
$
```

I have tried it on both laptop running ubuntu 20.04 and then RPi running ubuntu 21.04. Same behavior seen on both. Kindly check.

LuDuda, 2021-08-09:
Hello, @nageshshamnur any progress on your side?

I've just checked today (on two machines as well), and I can't reproduce this issue on both nRF52840 Dongle or DK (as i mentioned previously with MSD disabled). Let me paste my bash history, maybe we will find some differences. Please try to use the same revision of OT in both ot-daemon and RCP (however, your combination from above logs seems to work fine as well).

To clean-up repository:
```
$ cd ot-nrf528xx
$ sudo git clean -xdf
$ git fetch origin
$ git reset --hard ff68d16e32af5211d91591a0fe7806e904f34119

$ cd openthread
$ sudo git clean -xdf
$ git fetch origin
$ git reset --hard daf2ad7c245f947cadb2093af74ba9d121525473

$ cd ..
```

For nRF52840 Dongle:
```
$ rm -rf build
$ ./script/build nrf52840 USB_trans -DOT_BOOTLOADER=USB -DOT_THREAD_VERSION=1.2
$ arm-none-eabi-objcopy -O ihex build/bin/ot-rcp build/bin/ot-rcp.hex

# Flash the ot-rcp using nRF Connect for Desktop and nRF Programmer
```

For nRF52840 DK:
```
$ rm -rf build
$ script/build nrf52840 UART_trans -DOT_THREAD_VERSION=1.2
$ arm-none-eabi-objcopy -O ihex build/bin/ot-rcp build/bin/ot-rcp.hex
$ nrfjprog -f nrf52 --chiperase --program build/bin/ot-rcp.hex --reset

# Disable MSD - one-time operation per DK
$ JLinkExe -device NRF52840_XXAA -if SWD -speed 4000 -autoconnect 1
>MSDisable
>exit
# Power cycle the nRF52840 DK
```

Build `ot-daemon` and run it:
```
$ cd openthread
$ script/cmake-build posix -DOT_DAEMON=ON
$ sudo ./build/posix/src/posix/ot-daemon -v 'spinel+hdlc+uart:///dev/ttyACM0'

./build/posix/src/posix/ot-daemon[29716]: Running OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Aug  9 2021 22:31:43
./build/posix/src/posix/ot-daemon[29716]: Thread version: 3
./build/posix/src/posix/ot-daemon[29716]: Thread interface: wpan0
./build/posix/src/posix/ot-daemon[29716]: RCP version: OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; NRF52840; Aug  9 2021 22:30:13
```

Run `ot-ctl`:
```
sudo ./build/posix/src/posix/ot-ctl 
> version
OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; POSIX; Aug  9 2021 22:31:43
Done
> rcp version
OPENTHREAD/thread-reference-20200818-1092-gdaf2ad7c2; NRF52840; Aug  9 2021 22:30:13
Done
> dataset init new
Done
> dataset
Active Timestamp: 1
Channel: 20
Channel Mask: 0x07fff800
Ext PAN ID: 101cab5c4aaa8521
Mesh Local Prefix: fd43:da79:140e:5098::/64
Network Key: ec2a5a341ad3765c6bc7e228a36100dd
Network Name: OpenThread-1ec8
PAN ID: 0x1ec8
PSKc: fd7f0737f01b1631bd5a3055faa6b362
Security Policy: 672 onrcb
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
> state
leader
```

Some additional questions. Which version of nRF52840 Dongle and DK do you have?

If the problem still exists, could you connect the debugger or log what is the error message returned here: `src/posix/platform/radio.cpp:573`.

Also if you still see this error: `src/core/mac/sub_mac.cpp:521: void ot::Mac::SubMac::HandleTransmitDone(ot::Mac::TxFrame&, ot::Mac::RxFrame*, ot::Error): Assertion 'false' failed`, could you please log the error id that causes the assert?

jwhui, 2021-09-14:
Closing stale issue.

closed 2021-09-14 by jwhui

tunm1228, 2023-10-13:
> /opt/nrfjprog/

Why don't I have this folder "nrfjprog"?
[image]

jwhui, 2023-10-13:
> > /opt/nrfjprog/
> 
> Why don't I have this folder "nrfjprog"? [image]

@tunm1228 , you need to install [nRF Command Line Tools](https://www.nordicsemi.com/Products/Development-tools/nrf-command-line-tools/download)