openthread/ot-nrf528xx #137: Build NCP example for Uart - nrf52840

bhumikaajain, 2021-08-23:
Hi,

Can you please tell me how to rebuild nrf52840 NCP Uart example?
Here I need to modify the UART pins of nrf52840 hardware.
Default Uart Pins are GPIO 8 and GPIO 6.
In which file shall I update Uart pins and rebuild the example for my custom hardware which is based on chipset nrf52840.

edmont, 2021-08-24:
@bhumikaajain are you using nRF5 SDK or Nordic Connect SDK? Have you considered raising the issue in the [Nordic DevZone](https://devzone.nordicsemi.com/)?

bhumikaajain, 2021-08-24:
I am using NRF_SDK_for_thread_and_zigbee.
Also, I want to build it on the latest openthread, so in ot-nrf528xx, NCP UART example, if I want to built there, where shall I update UART pins.
I have written to nordic as well, but there is no update from their side yet.