openthread/ot-nrf528xx #139: Adding I2C(TWI) support in Openthread gives error while linking

Hard1kPrajapati, 2021-08-23:
Hello Team,

I am using Opnethread running on the nRF52840 device. I just want to add a few supports with my client application such as I2C, SPI, etc. But as soon as I add the code for I2C it gives me errors while linking.

**Environment setup**
I have cloned the new repo [OT-nRF52840](https://github.com/openthread/ot-nrf528xx) and using the thread client app(**openthread/examples/apps/cli/main.c**)

I have added the below code to the app to test first if it compiles:
```
#include "../../../../third_party/NordicSemiconductor/nrfx/drivers/include/nrfx_twi.h"

static nrfx_twi_t m_twi;
void twi_handler(nrfx_twi_evt_t const * p_event, void * p_context)
{
}
extern nrfx_err_t nrfx_twi_init(nrfx_twi_t const *        p_instance,
                         nrfx_twi_config_t const * p_config,
                         nrfx_twi_evt_handler_t    event_handler,
                         void *                    p_context);

void twi_init()
{
    const nrfx_twi_config_t twi_config = {
       .scl                = 26,
       .sda                = 27,
       .frequency          = 100,
       .interrupt_priority = 1,
       //.clear_bus_init     = false
    };
    m_twi.drv_inst_idx=0;
    nrfx_twi_init(&m_twi, &twi_config, twi_handler, NULL);
}
```
**Error I get**
```
main.c:(.text.twi_init+0x26): undefined reference to `nrfx_twi_init'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
I have also enabled the TWI and its instance in **sdk_config.h** file. Still, it shows the error.
I don't know where I am making the mistake, I wish to add the whole I2C code once it solves.

Thanks in advance,
Hardik

hubertmis, 2021-08-24:
Hi @Hard1kPrajapati ,

For complex applications running OpenThread on nRF52840 device with more subsystems enabled, I would recommend to use [nRF Connect SDK](https://www.nordicsemi.com/Products/Development-software/nRF-Connect-SDK) instead of OT-nRF52840. This SDK is available in Github: https://github.com/nrfconnect/sdk-nrf. In the SDK you get support for all peripherals available in nRF5 platform including I2C and SPI. I'm afraid enabling support for complex applications in OT-nRF52840 would require much effort.

As an alternative, you could have a look at [nRF5 SDK for Thread and Zigbee](https://www.nordicsemi.com/Products/Development-software/nRF5-SDK-for-Thread-and-Zigbee), an older but still maintained solution.

jwhui, 2022-03-25:
Closing stale issue.

closed 2022-03-25 by jwhui