openthread/ot-nrf528xx #148: Broken OT 1.2 support in openthread/ot-nrf528xx

George-Stefan, 2021-09-02:
It looks like for retransmissions of a frame with CSL IE and security enabled are corrupted.
Since the CSL IE phase is updated after the frame was secured.

See nrf_802154_tx_started() from https://github.com/openthread/ot-nrf528xx/blob/main/src/src/radio.c#L1216

edmont, 2021-09-03:
Hi @George-Stefan, thanks for spotting this. This PR should fix it: https://github.com/openthread/ot-nrf528xx/pull/149

Please notice that even after the fix the CSL IEs will be outdated for the retransmitted frames. This is a known issue and Thread Group is currently discussing a proper approach to address it.

edmont, 2021-09-03:
In any case OpenThread has a workaround already for the out of sync issue: https://github.com/openthread/openthread/pull/6342

closed 2021-09-04 by jwhui