openthread/ot-nrf528xx #170: Need guidance to build OpenThread correctly for my project

Totalus, 2021-10-01:
Hello,
I've been trying to compile OpenThread lately to replace the pre-compiled version that ships with the _nRF5 SDK for Thread and Zigbee_, but have yet been able to make a build that works properly.

Here are two compile configurations I have tried, both with different problems. I also shared how I linked the compiled libraries to the project.

## Compile configuration 1

Starting from the current master branch of this project. I add the following to the `ot-nrf528xx/src/nrf52840/openthread-core-nrf52840-config.h`

```
#ifndef OPENTHREAD_CONFIG_TMF_NETDATA_SERVICE_ENABLE
#define OPENTHREAD_CONFIG_TMF_NETDATA_SERVICE_ENABLE 1
#endif
```

Compile command : `./script/build nrf52840 UART_trans -DOT_COAP=1 -DOT_NETDATA_PUBLISHER=1 -DOT_BORDER_ROUTER=1`

### Problems with this config

I am getting a hard fault related to mbedtls.

[image]

Not sure what to do about that.

## Compile configuration 2

Same as configuration 1 but I also added the following in  `ot-nrf528xx/src/nrf52840/openthread-core-nrf52840-config.h`
```
#ifndef OPENTHREAD_CONFIG_CRYPTO_LIB
#define OPENTHREAD_CONFIG_CRYPTO_LIB OPENTHREAD_CONFIG_CRYPTO_LIB_PSA
#endif
```

### Problems with this configuration

The thread stack task is not running properly:

- `otTaskletsSignalPending` is called once, the task is executed
- `otSysEventSignalPending` is called once, the task is executed but the function `otSysProcessDrivers` seem to never return

<details>
<summary>Related code</summary>

```cpp
static void thread_stack_task(void * arg)
{
	UNUSED_PARAMETER(arg);
	while (1)
	{
		ASSERT(mp_ot_instance != NULL);
		otTaskletsProcess(mp_ot_instance);
		otSysProcessDrivers(mp_ot_instance); // Never returns
		UNUSED_RETURN_VALUE(ulTaskNotifyTake(pdTRUE, portMAX_DELAY));
	}
}

void otTaskletsSignalPending(otInstance * p_instance)
{
	if (thread_stack_task_handle == NULL)
		return;

	UNUSED_RETURN_VALUE(xTaskNotifyGive(thread_stack_task_handle));
}

void otSysEventSignalPending(void)
{
	static BaseType_t xHigherPriorityTaskWoken;
	if (thread_stack_task_handle == NULL)
		return;

	vTaskNotifyGiveFromISR(thread_stack_task_handle, &xHigherPriorityTaskWoken);
	portYIELD_FROM_ISR(xHigherPriorityTaskWoken);
}
```

</details>

## How I've linked the new compiled version my project

I've replaced the precompiled libraries files with the ones I compiled:

```sh
# Original OpenThread lib files in the project
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/lib/nrf52840/gcc/libopenthread-ftd.a
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/lib/nrf52840/gcc/libopenthread-nrf52840-transport.a
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/lib/nrf52840/gcc/libopenthread-platform-utils.a
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/lib/nrf52840/gcc/libopenthread-nrf52840-sdk.a
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/lib/nrf52840/gcc/libnordicsemi-nrf52840-radio-driver.a

# New paths pointing to the new compiled version (those are copied from ot-nrf528xx/build/lib)
MyProject/ThirdPartyLibraries/openthread/lib/libopenthread-ftd.a
MyProject/ThirdPartyLibraries/openthread/lib/libopenthread-nrf52840-transport.a
MyProject/ThirdPartyLibraries/openthread/lib/libopenthread-platform-utils-static.a
MyProject/ThirdPartyLibraries/openthread/lib/libopenthread-nrf52840-sdk.a
MyProject/ThirdPartyLibraries/openthread/lib/libnordicsemi-nrf52840-radio-driver.a
```

I've copied  `ot-nrf528xx/openthread/include` into  `ThirdPartyLibrary/openthread/include` and chaged **some** (not all) of the OpenThread related include path of my project:

```sh
# Original openthread related include paths
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/include
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/project/config
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/project/nrf52840

# New openthread related include path
MyProject/ThirdPartyLibraries/openthread/include # Changed
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/project/config   # Not change
MyProject/ThirdPartyLibraries/nRF5_SDK_for_Thread_and_Zigbee/external/openthread/project/nrf52840 # Not change
# I did not change the two last paths because I wasn't sure where to find the equivalent headers elsewhere...
```

Totalus, 2021-10-12:
Hello @LuDuda,
thanks for advising. I am trying to avoid moving an existing project to the _nRF Connect SDK_ indeed. If I can get this to work, it will save me the hassle.

I've replaced the `nrf_security` libs as well as you suggested, keeping the default `OPENTHREAD_CONFIG_CRYPTO_LIB_MBEDTLS` (Configuration # 1). This did solve the hard fault problem.

Now I have the same issue as Configuration # 2 : 

> The thread stack task is not running properly:
>
>    `otTaskletsSignalPending` is called once, the task is executed
>    `otSysEventSignalPending` is called once, the task is executed but the function `otSysProcessDrivers` seem to never return

Any idea on why I could be getting that behavior ?

Totalus, 2021-10-12:
Hey @LuDuda ,
I posted too fast. I realized that I've forgotten to copy the recompiled openthread lib files to my project. The stack seems to be running properly now. I'll see if it solves my problems with a few more tests.

To make it work I also had to tweek some function names in my code but also one line in the code from this repo, [this line](https://github.com/openthread/ot-nrf528xx/blob/d97253bb3372b0a4cc51fdd0c83c924500edba34/third_party/NordicSemiconductor/libraries/nrf_security/nrf_cc310_plat/src/nrf_cc3xx_platform_mutex_freertos.c#L106). I had to change `mutex->flag` to `mutext->flags`, a typo I believe.

LuDuda, 2021-10-12:
Good to hear!

Btw. typo is being addressed [here](https://github.com/nrfconnect/sdk-nrfxlib/pull/535/files), we will fetch changes when it is merged.

Should we close the issue for now?