openthread/ot-nrf528xx #172: `otbr-agent` fails to connect to nRF52840 dongle due to `RadioSpinelNoResponse`

fonix232, 2021-10-04:
This is my first foray into Thread/OpenThread, so please forgive possible confusions.

I've had a handful of nRF52840 dongles laying around (v1.2.0) and decided to give OT a quick go. Followed the instructions to set up the host service on a Raspberry Pi 4, which went without issues. Then followed this repo's [instructions](https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md#nrf52840-dongle-support-pca10059) as well, to create and flash the appropriate firmware for the dongle, which was also successful. 

The problem happens when I try to start `otbr-agent` - even though the proper TTY is selected, pretty soon after starting up, throws the error in the title. I've tried using different USB ports, swapping power supplies, to no avail. Same thing happens if I try to run the OTBR stack in Docker.

Steps to reproduce:

1. Build nRF52840 dongle RCP:

```
cd ~/ot-nrf528xx
./script/build nrf52840 USB_trans -DOT_BOOTLOADER=USB
```

2. Flash RCP firmware to dongle:

```
arm-none-eabi-objcopy -O ihex build/bin/ot-cli-ftd ot-cli-ftd.hex
nrfutil pkg generate --hw-version 52 --sd-req 0x00 --application-version 1 --application ot-cli-ftd.hex app_dfu_package.zip
nrfutil dfu usb-serial -pkg app_dfu_package.zip -p /dev/ttyACM0
```

3. Start OTBR in Docker:

```
docker run --sysctl "net.ipv6.conf.all.disable_ipv6=0 net.ipv4.conf.all.forwarding=1 net.ipv6.conf.all.forwarding=1" -p 8080:80 --dns=127.0.0.1 -it --volume /dev/ttyACM0:/dev/ttyACM0 --privileged openthread/otbr --radio-url spinel+hdlc+uart:///dev/ttyACM0
```

Result:

```
Oct  4 15:08:41 8d069c6d2888 otbr-agent[217]: [INFO]-UTILS---: Running 0.3.0-2af5cbd
Oct  4 15:08:41 8d069c6d2888 otbr-agent[217]: [INFO]-UTILS---: Thread version: 1.2.0
Oct  4 15:08:41 8d069c6d2888 otbr-agent[217]: [INFO]-UTILS---: Thread interface: wpan0
Oct  4 15:08:41 8d069c6d2888 otbr-agent[217]: [INFO]-UTILS---: Backbone interface: eth0
Oct  4 15:08:41 8d069c6d2888 otbr-agent[217]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Oct  4 15:08:41 8d069c6d2888 otbr-agent[217]: [WARN]-PLAT----: Error processing notification: Parse
Oct  4 15:08:41 8d069c6d2888 otbr-web[245]: [INFO]-WEB-----: Running 0.3.0-2af5cbd
Oct  4 15:08:41 8d069c6d2888 otbr-web[245]: [INFO]-WEB-----: Border router web started on wpan0
Oct  4 15:08:41 8d069c6d2888 otbr-web[245]: [ERR ]-WEB-----: OpenThread daemon is not running.
Oct  4 15:08:41 8d069c6d2888 otbr-agent[217]: [WARN]-PLAT----: Error processing notification: Parse
Oct  4 15:08:43 8d069c6d2888 otbr-agent[217]: [CRIT]-PLAT----: HandleRcpTimeout() at ../../third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2217: RadioSpinelNoResponse
```

The response is very similar to #71, however the solution won't work for me - the nRF52840 dongle does not have a J-Link OB debugger, and as such, it does not have MSD.

edmont, 2021-11-05:
Hi @fonix232, thanks for reporting this. Can you please state which commits are you using for `openthread` and `ot-nrf528xx` repos?

markus61, 2021-11-25:
hi,
same thing here. i am on:
Submodule path 'openthread': checked out '02d29cd5e343c3eb17840251b7feaa9345892895'

and:
commit 8853d0e9d1ce9471959dcfcf803beb49fae84030 (HEAD -> main, origin/main, origin/HEAD)
Author: dependabot[bot] <49699333+dependabot[bot]@users.noreply.github.com>
Date:   Wed Nov 24 08:20:05 2021 -0800

    submodule: bump openthread from `db9e430` to `02d29cd` (#203)
    
    Bumps [openthread](https://github.com/openthread/openthread) from `db9e430` to `02d29cd`.
    - [Commits](https://github.com/openthread/openthread/compare/db9e430876a50b179dd4a20f5159a4d77dfc81bb...02d29cd5e343c3eb17840251b7feaa9345892895)

markus61, 2021-11-25:
recipes found on net are wrong:
don't flash the cli, flash ot-rcp instead! 
arm-none-eabi-objcopy -O ihex build/bin/ot-cli-ftd ot-cli-ftd.hex needs to be
arm-none-eabi-objcopy -O ihex build/bin/ot-rcp ot-rcp.hex instead!

jwhui, 2021-11-25:
> recipes found on net are wrong: don't flash the cli, flash ot-rcp instead! arm-none-eabi-objcopy -O ihex build/bin/ot-cli-ftd ot-cli-ftd.hex needs to be arm-none-eabi-objcopy -O ihex build/bin/ot-rcp ot-rcp.hex instead!

@markus61 , can you point to the instructions you used so that we can make necessary updates?

pearson, 2021-12-20:
@jwhui , the confusion might come from the instructions [at this link](https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md#building-the-examples). Adding a few lines there to point out which build to use for which case might be useful, especially for those new to OpenThread.

fgervais, 2022-02-04:
I'm trying to do the same on my side. It seems like there is a lot of conflicting information right now with a mix of legacy and current stuff.

I think the best/current documentation for this right now is this:
https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/matter/openthread_rcp_nrf_dongle.html

Though I have yet to test it.

fgervais, 2022-02-05:
I'm here to report, the documentation mentioned in my previous post worked fine on my end.

jwhui, 2023-06-28:
Closing stale issue.

closed 2023-06-28 by jwhui