openthread/ot-nrf528xx #210: otbr-agent with NRF52840 (RCP) is exiting

saumildmehta, 2021-12-08:
I am using custom linux board and taken latest code of otbr-agent. Also taken latest code of ot-nrf528xx for NRF52840.

otbr-agent is able to communicate with RCP successfully and my openthread network is created as well. But randomly it files following error and exits:

    otbr-agent[14116]: 00:35:22.736 [WARN]-PLAT----: radio tx timeout otbr-agent[14116]: 00:35:22.736 [CRIT]-PLAT----: HandleRcpTimeout() at /usr/src/debug/otbr/git-r0/ot-br-posix/third_party/openthread/repo/src/lib/spinel /radio_spinel_impl.hpp:2218: RadioSpinelNoResponse

Full logs of otbr-agent from start are attached and it was exited without any activity. Once I was able to commission and communicate with device and after it got exited with same error. Is it issue from otbr? or rcp?

[otbr_agent_rcp_timeout.log](https://github.com/openthread/ot-nrf528xx/files/7675897/otbr_agent_rcp_timeout.log)

jwhui, 2021-12-09:
@saumildmehta , can you provide more information on your setup?

- Are you using an nRF52840-DK or USB dongle?
- How are you connecting the nRF52840 - UART, USB, I2C?

saumildmehta, 2021-12-13:
@jwhui Thanks for response.

**- What git commit(s) are you using?**

I am using following commits:
RCP: ot-nrf528xx - ad3b3c04bbe307253ff145a43287a2bd93eae3ef
OTBR: otbr - 83babaf236cad8471be28185d8d4351d37564919

**- Are you using an nRF52840-DK or USB dongle?**
**- How are you connecting the nRF52840 - UART, USB, I2C?**
I am using custom board with i.Mx6ull as host and NRF52840 module connected over UART.
UART between them is hardware flow control enabled.
Note: iMx6ull and NRF52840 connectivity is good as I have validated it by flashing different binary for BLE.

Let me know if you need any more logs or different trials.

jwhui, 2021-12-13:
@edmont @MarekPorwisz @LuDuda , have you seen this issue before?

saumildmehta, 2021-12-14:
@jwhui @edmont @MarekPorwisz @LuDuda 
FYI. If I flash image first time, it runs for long time may be for couple of hours and able to add and ping child devices.
But after words it exits in seconds. Majority times I have seen this "radio tx timeout" error after applying command "ot-ctl thread start".

LuDuda, 2021-12-17:
@saumildmehta are you able to connect debugger to the nRF52840 SoC after this error happens? It would be interesting to see what is the state of the device (maybe it hangs in some assert).

Could you verify if you can see the same problem with nRF52840 DK that acts as RCP? (If you use J-Link USB, remember to turn off the MSD as it is described in the platform's README.md).

LancerSky, 2022-05-16:
We are developing OTBR base on OpenWrt. And I also got the same exception in version https://github.com/openthread/ot-br-posix/commit/5be35949c6f1fd4cc3a8e63169cec6a2211135a4.
The EFR32MG21 RCP firmware version was built from https://github.com/openthread/ot-efr32/commit/31e9fc8606d9a68cd79868d2b3169992168a7398.
The end device firmware version was built from ot-nrf52840 - https://github.com/openthread/ot-nrf528xx/commit/6533ff751f1c5e05ca48bda15f7b7bc3419fc8e2
```
Thu May 12 05:47:09 2022 user.info otbr-agent[3232]: 00:06:24.745 [INFO]-MAC-----: Received IPv6 HopOpts msg, len:83, chksum:0000, from:0x7000, sec:yes, prio:normal, rss:-40.0
Thu May 12 05:47:09 2022 user.info otbr-agent[3232]: 00:06:24.745 [INFO]-MAC-----:     src:[fdd2:b178:8ba5:0:6600:5182:e6dd:13f6]
Thu May 12 05:47:09 2022 user.info otbr-agent[3232]: 00:06:24.745 [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.991 [DEBG]-MAC-----: Received frame from short address 0x7000
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: Rx security - frame counter 37487
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: =========================[RX seqno=189 len=078]=========================
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: | 49 98 BD 32 3D FF FF 00 | 70 0D 6F 92 00 00 01 7C | I.=2=...p.o....|
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: | 5A 3F 50 F1 D7 EF 43 C2 | 4D D3 03 00 00 01 E1 06 | Z?PqWoCBMS....a.
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: | 6D 04 40 E2 70 01 F0 9C | 95 16 33 D8 2C 58 03 82 | m.@bp.p...3X,X..
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.993 [DEBG]-MAC-----: | 0B 76 12 4A 10 5F 73 2C | FC BC 6E 65 74 77 6F 72 | .v.J._s,|<networ
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.993 [DEBG]-MAC-----: | 6B 5F 74 65 73 74 FF 14 | AB DD 77 4F 66 10 .. .. | k_test..+]wOf...
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.993 [DEBG]-MAC-----: ------------------------------------------------------------------------
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:24.993 [INFO]-MAC-----: Received IPv6 HopOpts msg, len:83, chksum:0000, from:0x7000, sec:yes, prio:normal, rss:-40.0
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:24.993 [INFO]-MAC-----:     src:[fdd2:b178:8ba5:0:50f1:d7ef:43c2:4dd3]
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:24.993 [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.038 [DEBG]-MAC-----: Received frame from short address 0x7000
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.039 [DEBG]-MAC-----: Rx security - frame counter 37488
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.041 [DEBG]-MAC-----: =========================[RX seqno=190 len=078]=========================
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.041 [DEBG]-MAC-----: | 49 98 BE 32 3D FF FF 00 | 70 0D 70 92 00 00 01 7C | I.>2=...p.p....|
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.041 [DEBG]-MAC-----: | 5A 3F 50 F1 D7 EF 43 C2 | 4D D3 03 00 00 01 E1 06 | Z?PqWoCBMS....a.
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.043 [DEBG]-MAC-----: | 6D 04 40 E2 70 01 F0 9C | 95 16 33 D8 2C 58 03 82 | m.@bp.p...3X,X..
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.043 [DEBG]-MAC-----: | 0B 76 12 4A 10 5F 73 2C | FC BC 6E 65 74 77 6F 72 | .v.J._s,|<networ
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.043 [DEBG]-MAC-----: | 6B 5F 74 65 73 74 FF 14 | 2E B2 9B 6F 66 10 .. .. | k_test...2.of...
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.044 [DEBG]-MAC-----: ------------------------------------------------------------------------
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:25.044 [INFO]-MAC-----: Received IPv6 HopOpts msg, len:83, chksum:0000, from:0x7000, sec:yes, prio:normal, rss:-40.0
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:25.045 [INFO]-MAC-----:     src:[fdd2:b178:8ba5:0:50f1:d7ef:43c2:4dd3]
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:25.045 [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
Thu May 12 05:47:10 2022 user.warn otbr-agent[3232]: 00:06:25.046 [WARN]-PLAT----: radio tx timeout
Thu May 12 05:47:10 2022 user.warn otbr-agent[3232]: 00:06:25.046 [WARN]-PLAT----: RCP failure detected
Thu May 12 05:47:10 2022 user.warn otbr-agent[3232]: 00:06:25.046 [WARN]-PLAT----: Trying to recover (1/5)
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.047 [DEBG]-PLAT----: Wait response: tid=0 key=0
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:25.083 [INFO]-PLAT----: RCP reset: RESET_POWER_ON
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.083 [DEBG]-PLAT----: Wait response: tid=1 key=32
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.084 [DEBG]-PLAT----: Wait response: tid=2 key=54
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.120 [DEBG]-PLAT----: Wait response: tid=3 key=53
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.125 [DEBG]-PLAT----: Wait response: tid=4 key=52
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.160 [DEBG]-PLAT----: Wait response: tid=5 key=33
[... cut ...]
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-ADPROXY-: Started
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-DPROXY--: started
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-BA------: Start Thread Border Agent: OK
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-UBUS----: Connected as 81469bc6
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-UBUS----: Uloop run
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-APP-----: Border router agent started.
```

kroa38, 2023-01-25:
Hello,
My OTBR is based on the Silabs Image for Raspberry-Pi 4 [here](https://github.com/SiliconLabs/matter/blob/release_1.0.2-1.0/docs/silabs/general/ARTIFACTS.md)  and when i attache the nRF52840 DK to Rpi the service
"otbr-agent" return an error because the RCP is in version 8 and not in the range of [4-6]  (Spinel issue i think)
Is there a way to downgrade to a RCP version 4 to 6 ?

jwhui, 2023-01-25:
> Hello, My OTBR is based on the Silabs Image for Raspberry-Pi 4 [here](https://github.com/SiliconLabs/matter/blob/release_1.0.2-1.0/docs/silabs/general/ARTIFACTS.md) and when i attache the nRF52840 [...]

@kroa38 , you can move to an earlier commit. The easiest is to match the same commit as what's used on the OTBR host.

kroa38, 2023-01-26:
Any idea how to know the right commit with a RCP API version in the range of [4-6]
Thanks

jwhui, 2023-01-26:
> Any idea how to know the right commit with a RCP API version in the range of [4-6] Thanks

As mentioned above, one method is to simply use the same git commit on both OTBR host and RCP.

Alternatively, you can walk back the git blame history to find when `SPINEL_RCP_API_VERSION` was set to 6.

https://github.com/openthread/openthread/blob/9af0bfa60e373d81a5576b298d6664045870a375/src/lib/spinel/spinel.h#L420

closed 2023-02-06 by jwhui

MonicaisHer, 2023-10-05:
> Any idea how to know the right commit with a RCP API version in the range of [4-6] Thanks

Here is my solution: Building and flashing the RCP using [main repo commit](https://github.com/openthread/ot-nrf528xx/commit/00ac6cd0137a4f09288b455bf8d7aa72d74062d1) (00ac6cd0137a4f09288b455bf8d7aa72d74062d1) and [submodule commit](https://github.com/openthread/openthread/blob/9af0bfa60e373d81a5576b298d6664045870a375) (9af0bfa60e373d81a5576b298d6664045870a375). By doing this, the RCP and OT-BR will be compatible using the same API version 6.