openthread/ot-nrf528xx #213: Compile failed

LancerSky, 2021-12-13:
Hi,
I want to build an ot-cli-ftd firmware that can automatically join the Thread Network.
So, I use the lastest source code and added the code as following, 
```
diff --git a/examples/apps/cli/main.c b/examples/apps/cli/main.c
index 4d0440c12..39171bab3 100644
--- a/examples/apps/cli/main.c
+++ b/examples/apps/cli/main.c
@@ -130,6 +130,9 @@ pseudo_reset:
 
     otAppCliInit(instance);
 
+    otIp6SetEnabled(instance, true);
+    otThreadSetEnabled(instance, true);
+    
 #if OPENTHREAD_POSIX && !defined(FUZZING_BUILD_MODE_UNSAFE_FOR_PRODUCTION)
     otCliSetUserCommands(kCommands, OT_ARRAY_LENGTH(kCommands), instance);
 #endif
```
But the compiler generates an error:
```main.c:(.text.startup.main+0x1c): undefined reference to `otIp6SetEnabled'```
Here is the full log. 
[compile error log.txt](https://github.com/openthread/ot-nrf528xx/files/7702872/compile.error.log.txt)
Is there anybody to help me with this issue? Thanks.

jwhui, 2021-12-13:
Did you add `#include <openthread/ip6.h>` to the top of the file?

LancerSky, 2021-12-14:
Yeah, I also tested this. But the same error.
```
diff --git a/examples/apps/cli/main.c b/examples/apps/cli/main.c
index 4d0440c12..b156a7ae4 100644
--- a/examples/apps/cli/main.c
+++ b/examples/apps/cli/main.c
@@ -33,6 +33,7 @@
 #include <openthread/cli.h>
 #include <openthread/diag.h>
 #include <openthread/tasklet.h>
+#include <openthread/ip6.h>
 #include <openthread/platform/logging.h>
 
 #include "openthread-system.h"
@@ -130,6 +131,9 @@ pseudo_reset:
 
     otAppCliInit(instance);
 
+    otIp6SetEnabled(instance, true);
+    otThreadSetEnabled(instance, true);
+    
 #if OPENTHREAD_POSIX && !defined(FUZZING_BUILD_MODE_UNSAFE_FOR_PRODUCTION)
     otCliSetUserCommands(kCommands, OT_ARRAY_LENGTH(kCommands), instance);
 #endif
```

closed 2021-12-14 by LancerSky