openthread/ot-nrf528xx #223: Cannot connect to existing network

TPolzer, 2021-12-28:
I have an existing Thread network consisting of several openthread nodes (with different versions). I specifically (in isolation) tested against 0247d02b9dbfb437a1b8e81040323937814e0aab (from 2021-01-15).

I tried adding a new node with this new separate repository:  
I built ot-cli-ftd with `./script/build nrf52840 USB_trans -DOT_BOOTLOADER=USB` from 5a926809e0df6c506fc0dceb8998409405b7ece7.

I can kind of connect to the existing network, but not fully. Following https://github.com/openthread/openthread/blob/main/src/cli/README_DATASET.md#attach-to-existing-network
- after connecting, `dataset active` only lists the "Network Key:", nothing else
- `state` reports "child" (`routereligible` says "Enabled").
- It uses the wrong mesh local prefix: the new node uses `fdde:ad00:beef:0:0:ff:fe00:1c01`, while its parent is `fdbf:e492:40f3:de5a:0:ff:fe00:1c00`.
- Consequently, pinging with `ff02::1` works in both directions, `ff03::1` doesn't.

Let me know if this bug report should be moved to the openthread top level repository. I don't have any other hardware to test against.

TPolzer, 2021-12-28:
I've also tried with existing dataset in the settings flash:
- I flash ot-cli-ftd from 0247d02b9dbfb437a1b8e81040323937814e0aab
- I attach to the existing network
- I flash the new build from 5a926809e0df6c506fc0dceb8998409405b7ece7
- Now `dataset active` reports the correct network details, including the correct mesh local prefix
- `ipaddr` still shows the (wrong) `fdde:ad00:beef::/64" prefix
- The node doesn't actually manage to connect to the existing network, it becomes a "leader" with no neighbors.

jwhui, 2022-01-05:
@TPolzer , can you provide the output of `dataset active` for your existing Thread network? In particular, what is the Active Timestamp value? In order for the [Attach to Existing Network](https://github.com/openthread/openthread/blob/main/src/cli/README_DATASET.md#attach-to-existing-network) steps to succeed, the Active Timestamp of the existing network must be non-zero.

TPolzer, 2022-01-08:
With the old (working) version:
```
> version
OPENTHREAD/20191113-01630-g0247d02b9; NRF52840; Dec 27 2021 20:42:25
Done
> dataset active
Active Timestamp: 1
Channel: 26
Channel Mask: 0x07fff800
Ext PAN ID: c9c3d1247c9a448b
Mesh Local Prefix: fdbf:e492:40f3:de5a::/64
Master Key: ***secret***
Network Name: OpenThread-e217
PAN ID: 0xe217
PSKc: **secret?**
Security Policy: 0, onrcb
Done
> state
child
Done
> neighbor table
| Role | RLOC16 | Age | Avg RSSI | Last RSSI |R|D|N| Extended MAC     |
+------+--------+-----+----------+-----------+-+-+-+------------------+
|   R  | 0x2400 |   0 |      -73 |       -75 |1|1|1| baaa5854cf8d3407 |
|   R  | 0x4400 |  10 |      -80 |       -79 |1|1|1| 5e1f05066d0b9892 |
|   R  | 0x7c00 |  11 |      -73 |       -73 |1|1|1| 4e049f5bee138a2b |

Done
> ipaddr
2a02:**secret**
fdbf:e492:40f3:de5a:0:ff:fe00:ac09
fdbf:e492:40f3:de5a:77ca:28f2:675f:dec4
fe80:0:0:0:d839:76a9:816f:6cdf
Done
> ping ff02::1
Done
> 16 bytes from fe80:0:0:0:ccd9:6984:18e9:3d2b: icmp_seq=1 hlim=64 time=12ms

> ping ff03::1
Done
> 16 bytes from fdbf:e492:40f3:de5a:cd49:ed25:8df8:553e: icmp_seq=2 hlim=64 time=11ms
16 bytes from fdbf:e492:40f3:de5a:8195:2838:718e:72ab: icmp_seq=2 hlim=64 time=376ms
16 bytes from fdbf:e492:40f3:de5a:f75f:990d:102e:b00d: icmp_seq=2 hlim=64 time=387ms
16 bytes from fdbf:e492:40f3:de5a:2fe4:3747:c157:377c: icmp_seq=2 hlim=64 time=393ms
16 bytes from fdbf:e492:40f3:de5a:8bfb:a3fa:d493:579b: icmp_seq=2 hlim=64 time=401ms
16 bytes from fdbf:e492:40f3:de5a:a5b2:d15a:3b7c:7b9: icmp_seq=2 hlim=63 time=406ms
```

Flashing the new version (without erasing settings):
```
> version
OPENTHREAD/thread-reference-20200818-1436-g442b7e82f; NRF52840; Jan  8 2022 12:33:45
Done
> dataset active
Active Timestamp: 1
Channel: 26
Channel Mask: 0x07fff800
Ext PAN ID: c9c3d1247c9a448b
Mesh Local Prefix: fdbf:e492:40f3:de5a::/64
Network Key: ***secret, correct key***
Network Name: OpenThread-e217
PAN ID: 0xe217
PSKc: ***secret, correct key***
Security Policy: 0 onrcb
Done
> state
leader
Done
> neighbor table
| Role | RLOC16 | Age | Avg RSSI | Last RSSI |R|D|N| Extended MAC     |
+------+--------+-----+----------+-----------+-+-+-+------------------+

Done
> factoryreset
> dataset networkkey **secret**
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
> state
child
Done
> neighbor table
| Role | RLOC16 | Age | Avg RSSI | Last RSSI |R|D|N| Extended MAC     |
+------+--------+-----+----------+-----------+-+-+-+------------------+
|   R  | 0x0400 |  23 |      -60 |       -60 |1|1|1| a691d5d5fa1d1838 |
|   R  | 0x1400 |  24 |      -84 |       -83 |1|1|1| 9ea1180198677fef |
|   R  | 0x7c00 |  22 |      -76 |       -74 |1|1|1| 4e049f5bee138a2b |

Done
> dataset active
Network Key: **secret, correct key**
Done
> ipaddr
2a02:**secret, with correct public prefix**
fdde:ad00:beef:0:0:ff:fe00:ac0a
fdde:ad00:beef:0:503e:ccc4:3e81:7db6
fe80:0:0:0:24ac:78a8:b601:3c7f
Done
> ping ff02::1
16 bytes from fe80:0:0:0:ccd9:6984:18e9:3d2b: icmp_seq=1 hlim=64 time=11ms
1 packets transmitted, 1 packets received. Round-trip min/avg/max = 11/11.0/11 ms.
Done
> ping ff03::1
1 packets transmitted, 0 packets received.
Done
```

TPolzer, 2022-01-08:
Flashing two dongles with the new openthread version and a fresh dataset (`dataset init new` on one of them) works btw.

TPolzer, 2022-01-08:
I'm wondering whether it has something to do with timesync being enabled in the existing network. Though
- my understanding is attaching as a child should still work 
- I tried compiling the new openthread with `-DOT_TIME_SYNC=true`, the observed behavior (see https://github.com/openthread/ot-nrf528xx/issues/223#issuecomment-1007970471) is identical, time sync works:
```
> version
OPENTHREAD/thread-reference-20200818-1436-g442b7e82f; NRF52840; Jan  8 2022 12:33:45
Done
> networktime
Network Time:     1654448362us (synchronized)
Time Sync Period: 30s
XTAL Threshold:   300ppm
Done
> ping ff02::1
16 bytes from fe80:0:0:0:ccd9:6984:18e9:3d2b: icmp_seq=4 hlim=64 time=12ms
1 packets transmitted, 1 packets received. Round-trip min/avg/max = 12/12.0/12 ms.
Done
> ping ff03::1
1 packets transmitted, 0 packets received.
Done
```

jwhui, 2022-01-15:
This seems related to the issue reported in this [openthread-users email](https://groups.google.com/g/openthread-users/c/UsgEMScMQlQ/m/wYKu-plXBQAJ).

I have submitted a proposed fix in https://github.com/openthread/openthread/pull/7323. Let me know if this addresses your issue.

TPolzer, 2022-01-15:
Works with the pull request. Thanks!

jwhui, 2022-01-18:
> Works with the pull request. Thanks!

@TPolzer , thanks for testing and reporting back! https://github.com/openthread/openthread/pull/7323 has been merged.

closed 2022-01-18 by jwhui