openthread/ot-nrf528xx #227: builds from 8bed800 have no radio connectivity

jahess, 2022-01-03:
Specifically I have 2 nrf52840 dongles that have no connectivity with each other or a pair of efr32mg12/BRD4166A's.

Using arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release) on both MacOSX 12.1 and RedHat 7.something...

rm -rf build
script/build nrf52840 USB_trans -DOT_JOINER=ON -DOT_COMMISSIONER=ON -DOT_BOOTLOADER=USB
arm-none-eabi-objcopy -O ihex build/bin/ot-cli-ftd build/bin/ot-cli-ftd.hex
nrfutil pkg generate --hw-version 52 --sd-req=0x00 --application build/bin/ot-cli-ftd.hex --application-version 1 build/bin/ot-cli-ftd.zip
nrfutil dfu usb-serial -pkg build/bin/ot-cli-ftd.zip -p /dev/tty.<macosx/linux specific>

All result in no apparent error.  After flashing, I'm able to screen into both devices.  Using steps from:

    https://openthread.io/codelabs/openthread-hardware#0

namely:

dataset init new
dataset commit active
ifconfig up
thread start

Both report no issue at every step.  Then I:

scan

and neither device ever reports anything found (ie each other) during the scan.

FWIW, I gave up on the nrf dongles and tried the same (only similar build steps) on the efr32's and... they both will scan and show each other.  With them up, I re-tried the scan's on the nrf's and they still show nothing.

No idea what to try next.  Is there a build and test daemon with hardware somewhere that I can check a web page on to convince me 8bed800 isn't totally broken?

TPolzer, 2022-01-04:
Sounds similar to my experience on issue #223.

jahess, 2022-01-05:
Yes it does.  Do you have the non-dongle development boards?  Do they connect more... robustly?

If we knew, it might help in starting to narrow things down.

jwhui, 2022-01-07:
@jahess , I just tried on two Nordic nRF52840 USB dongles and everything worked as expected.

```
git checkout 8bed800
git submodule update
script/build nrf52840 USB_trans -DOT_JOINER=ON -DOT_COMMISSIONER=ON -DOT_BOOTLOADER=USB
arm-none-eabi-objcopy -O ihex build/bin/ot-cli-ftd build/bin/ot-cli-ftd.hex
```

I used the nRF Connect for Desktop programmer to flash the image onto two nRF52840 USB dongles.

On Node 1:
```
> version
OPENTHREAD/thread-reference-20200818-1416-ga65afcdbb; NRF52840; Jan  7 2022 13:14:28
Done
> dataset init new
Done
> dataset commit active
Done
> dataset active
Active Timestamp: 1
Channel: 13
Channel Mask: 0x07fff800
Ext PAN ID: 380f58f08a921616
Mesh Local Prefix: fdb7:99ff:530c:148d::/64
Network Key: 8b315ef2fb87e16c81ef079218f0518c
Network Name: OpenThread-4f92
PAN ID: 0x4f92
PSKc: 9b023224e65e163400ed3ca8be0ea641
Security Policy: 672 onrcb
Done
> ifconfig up
Done
> thread start
Done
```

On Node 2:
```
> version   
OPENTHREAD/thread-reference-20200818-1416-ga65afcdbb; NRF52840; Jan  7 2022 13:14:28
Done
> dataset init new
Done
> dataset commit active
Done
> dataset active
Active Timestamp: 1
Channel: 18
Channel Mask: 0x07fff800
Ext PAN ID: 301b33fd1cd44def
Mesh Local Prefix: fdf6:9380:166b:d13a::/64
Network Key: a8ac4e6fd70f761496e857b55944a324
Network Name: OpenThread-9c5b
PAN ID: 0x9c5b
PSKc: 1076411a91176bf0b386490118614ec9
Security Policy: 672 onrcb
Done
> ifconfig up
Done
> thread start
Done
```

Scan on Node 1:
```
> scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 | OpenThread-9c5b  | 301b33fd1cd44def | 9c5b | 662313a11970b1e6 | 18 | -31 | 252 |
Done
```

Scan on Node 2:
```
> scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 | OpenThread-4f92  | 380f58f08a921616 | 4f92 | 425e73cb55f85266 | 13 | -28 | 255 |
Done
```

@LuDuda , any ideas?

jahess, 2022-01-08:
Great to hear.  Could you tell me what cross compiler version you were using?

jwhui, 2022-01-08:
> Great to hear. Could you tell me what cross compiler version you were using?

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
```

jwhui, 2022-03-22:
Closing stale issue.

closed 2022-03-22 by jwhui