openthread/ot-nrf528xx #237: UART Abstraction hangs in otPlatUartFlush

TPolzer, 2022-01-15:
I have an nrf52840 dongle, flashed to ot-cli-ftd from head (5b0429ecf6cb4837080d8adb6aceb8c5eb2fc026).

Right after reset, if I execute `router table` eleven times, it hangs in `otPlatUartFlush`, with `(sUsbState.mTransferInProgress && !sUsbState.mTransferDone) == true` indefinitely.

When I connect to a network and print some tables it hangs a lot faster, so I guess it's in some way proportional to the amount of output?

jwhui, 2022-01-18:
@TPolzer , thanks for raising this issue. I was able to reproduce your issue using the latest main branch (856c280bed73ba8846ded933ab4cf758e9af2dae).

@MarekPorwisz , can you help take a look?

TPolzer, 2022-02-06:
If I insert a call to `nrf5UartProcess()` inside the loop in `otPlatUartFlush` (it clearly needs to be there), it prints a spurious newline instead of getting stuck. No idea where that's coming from, maybe that's actually a bug in the cli example?

jwhui, 2022-02-07:
@LuDuda @MarekPorwisz , can someone help with this issue?

canisLupus1313, 2022-02-08:
@TPolzer Looks like the events on usb bus aren't handled/triggered correctly when we are near to end of capacity of `sTxBuffer` seems that incorrect params are passed to TX function.

The reason that adding `nrf5UartProcess` helps is because that function calls `app_usbd_event_queue_process` which triggers usbd events resumes the transfer.

Never the less I'm trying to figure out what is happening.

jwhui, 2022-03-22:
Resolved by #255

closed 2022-03-22 by jwhui