openthread/ot-nrf528xx #257: Noob - Build options for OTBR RCP build

HeneryH, 2022-02-10:
There seem to be different ways to build such as west build, script build and make.

The instructions in the readme call for 
`$ ./script/build nrf52840 USB_trans`

However, the OT Border Router building instructions say that I need to build an RCP Platform flavor of the nrf52840dk and it should be built with the following options
`BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 USB=1 LINK_RAW=1`

I'm not sure how to get from here to there?

Any help would be greatly appreciated.

I could not build on my rpi due to tool limitations and I couldn't figure out how to get the build option done with Visual Code so I am using my WSL2 Ubuntu machine and have done the following setup steps on the WSL2 instance.  But I still couldn't get there.

```
# I got the kitware sh
sudo bash kitware-archive.sh
sudo apt install --no-install-recommends git cmake ninja-build gperf \\n  ccache dfu-util device-tree-compiler wget \\n  python3-dev python3-pip python3-setuptools python3-tk python3-wheel xz-utils file \\n  make gcc gcc-multilib g++-multilib libsdl2-dev
cmake --version
python3 --version
dtc --version
# I got the gn binary and added to path
vi ${HOME}/.bashrc
pip3 install --user west
ls ~/.local
vi ~/.bashrc # to add the .local/bin to the path
source ${HOME}/.bashrc
cd ~/ncs
west init -m https://github.com/nrfconnect/sdk-nrf --mr main
west update
west zephyr-export
pip3 install --user -r zephyr/scripts/requirements.txt
pip3 install --user -r nrf/scripts/requirements.txt
pip3 install --user -r bootloader/mcuboot/scripts/requirements.txt
mkdir ~/gnuarmemb # and get the bz2 file
cd ~/gnuarmemb
tar -xvjf  ../gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2
vi ~/.bashrc # to add the environment variables
cd ~/ncs
source zephyr/zephyr-env.sh
cd ~/openthread
./script/bootstrap
./bootstrap
cd ~
git clone git@github.com:openthread/ot-nrf528xx.git # the new location of the nrf files
cd ot-nrf528xx
git submodule update --init
./script/bootstrap
./script/build nrf52840 USB_trans  # [this works, the following doesn't] 
./script/build nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 USB=1 LINK_RAW=1
```

jwhui, 2022-02-10:
To build RCP for nRF52840 direct from this GitHub repo, you can follow the steps in [src/nrf52840/README.md](https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md).

Prepare the repo:
```
$ git clone https://github.com/openthread/ot-nrf528xx
$ cd ot-nrf528xx
$ git submodule update --init
$ ./script/bootstrap
```

For nRF52840-DK:
```
$ ./script/build nrf52840 USB_trans
```

For nRF52840 USB Dongle:
```
$ ./script/build nrf52840 USB_trans -DOT_BOOTLOADER=USB
```

HeneryH, 2022-02-10:
The [https://openthread.io/platforms/co-processor/firmware](Instructions for building RCP) call for using the following build parameters that I just found out the correct syntax for.

The following works with the OTBR as of today.
```
./script/build nrf52840 USB_trans -DOT_BORDER_AGENT=ON -DOT_BORDER_ROUTER=ON -DOT_COMMISSIONER=ON -DOT_UDP_FORWARD=ON -DOT_LINK_RAW=ON

arm-none-eabi-objcopy -O ihex build/bin/ot-rcp build/bin/ot-rcp.hex
```

jwhui, 2022-02-10:
> Can this perhaps be a documentation enhancement?

Yes, we'll work to update the documentation. Thanks for pointing this out!

Vyrastas, 2022-02-11:
We have an internal bug logged to work on this.

jwhui, 2023-06-28:
Closing stale issue.

closed 2023-06-28 by jwhui