openthread/ot-nrf528xx #296: Nordic nRF52840DK - Too many RCP failures while running ot-daemon

marcorussi, 2022-04-06:
Hello all,
I have been trying to run a RCP on my nRF52840DK board for days with no success. I have also tried many commits from this repo and the sample provided by Nordic in their nRF Connect SDK with "partial" success.
Although I am going to raise an issue regarding the latest commit of this repo (ot-nrf52840) which represents the main issue I am facing at the moment. 
I am on main branch commit id 83f2517 with openthread git submodule commit id 7dfde1f.
I have built openthread library (the git submodule) and the nRF52840 images as below:
openthread: 
`./script/cmake-build posix -DOT_THREAD_VERSION=1.2 -DOT_DAEMON=ON
`
nRF52840
`./script/build nrf52840 UART_trans`
and
```
cd build/bin
arm-none-eabi-objcopy -O ihex ot-cli-ftd ot-cli-ftd.hex
```
I have flashed my nRF52840DK (PCA10056 / V0.9.0 / 2017.10):
`nrfjprog -f nrf52 --chiperase --verify --program ot-cli-ftd.hex --reset`
and then disable MSD and UART HW flow control check on my board with the _MSDDisable_ and _SetHWFC Force_ JLink commands.

Finally I launched the OT daemon:
_./openthread/build/posix/src/posix/ot-daemon 'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200?uart-flow-control' --verbose_

Here the result:
_/openthread/build/posix/src/posix/ot-daemon[17922]: Running OPENTHREAD/thread-reference-20200818-1609-g7dfde1f12; POSIX; Apr  6 2022 10:40:02
./openthread/build/posix/src/posix/ot-daemon[17922]: Thread version: 3
./openthread/build/posix/src/posix/ot-daemon[17922]: 49d.17:48:13.881 [C] Platform------: Too many rcp failures, exiting
./openthread/build/posix/src/posix/ot-daemon[17922]: 49d.17:48:13.881 [C] Platform------: BSD TCP function() at radio_spinel_impl.hpp:2259: Failure_

Thank you.

marcorussi, 2022-04-06:
I found a fix about removing a zombie wpan0 but it seems that I don't have it:
```
marco@marco-Latitude-5590:~/Pingeco/exp/ot-nrf528xx$ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: enp0s31f6: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether 8c:ec:4b:e5:90:08 brd ff:ff:ff:ff:ff:ff
3: wlp2s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether a0:88:69:f5:22:9c brd ff:ff:ff:ff:ff:ff
4: nat64: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP mode DEFAULT group default qlen 500
    link/none 
5: otbr0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default 
    link/ether 02:42:12:ee:8d:eb brd ff:ff:ff:ff:ff:ff
6: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default 
    link/ether 02:42:6a:2d:01:20 brd ff:ff:ff:ff:ff:ff
```

jwhui, 2022-04-06:
@marcorussi , thanks for raising this issue. I believe you are experiencing the same issue as in https://github.com/openthread/openthread/issues/7552, which should be addressed by https://github.com/openthread/openthread/pull/7570.

marcorussi, 2022-04-07:
Hi jwhui, that is what I thought as I commented on that issue before opening this one. The only thing that confuses me is the "Unexpected RCP reset: RESET_POWER_ON" message. Do you think it has to do with the same TCP error? If yes, should I try commit 58e4f85749e9b2cba72d4aacecf62655b3d167d5 ?

canisLupus1313, 2022-04-07:
@marcorussi Looks like You have problem when starting the RCP. Would it be possible for You to provide dmesg log? Also can You check `lsusb` for nRF dk presence? 

And one more thing please meake sure that when You preform git operations on repo You always remove the build folder.

marcorussi, 2022-04-07:
@canisLupus1313 please find the dmesg log. I took the only part I think you'd need:
```
[  658.390253] usb 1-3: USB disconnect, device number 8
[  664.366472] usb 1-9: new high-speed USB device number 11 using xhci_hcd
[  664.515121] usb 1-9: New USB device found, idVendor=1366, idProduct=0105, bcdDevice= 1.00
[  664.515126] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  664.515128] usb 1-9: Product: J-Link
[  664.515130] usb 1-9: Manufacturer: SEGGER
[  664.515132] usb 1-9: SerialNumber: 000683274154
[  664.517095] cdc_acm 1-9:1.0: ttyACM0: USB ACM device
[  873.142137] usb 1-9: USB disconnect, device number 11
[  876.319043] usb 1-9: new high-speed USB device number 12 using xhci_hcd
[  876.468307] usb 1-9: New USB device found, idVendor=1366, idProduct=0105, bcdDevice= 1.00
[  876.468325] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  876.468331] usb 1-9: Product: J-Link
[  876.468336] usb 1-9: Manufacturer: SEGGER
[  876.468341] usb 1-9: SerialNumber: 000683274154
[  876.471153] cdc_acm 1-9:1.0: ttyACM0: USB ACM device
```
And yes, the device is present as you can see. Find also here lsusb:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 1366:0105 SEGGER J-Link
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 002: ID 0bda:5650 Realtek Semiconductor Corp. Integrated Webcam_HD
Bus 001 Device 010: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 001 Device 004: ID 0a5c:5832 Broadcom Corp. 5880
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Thank you.

marcorussi, 2022-04-07:
I have updated openthread submodule to [a9b4e637f](https://github.com/openthread/openthread/commit/a9b4e637f930e1a295d8292b6c16055c19eba23a)
and rebuilt ot-rcp nrf52840 image and ot-daemon/ot-ctl. 
It fails with a different error now:
```
marco@marco-Latitude-5590:~/Pingeco/exp/ot-nrf528xx$ sudo ./openthread/build/posix/src/posix/ot-daemon 'spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=115200' --verbose
./openthread/build/posix/src/posix/ot-daemon[21864]: Running OPENTHREAD/thread-reference-20200818-1621-ga9b4e637f; POSIX; Apr  7 2022 22:11:14
./openthread/build/posix/src/posix/ot-daemon[21864]: Thread version: 3
./openthread/build/posix/src/posix/ot-daemon[21864]: 49d.17:40:41.436 [C] Platform------: Too many rcp failures, exiting
./openthread/build/posix/src/posix/ot-daemon[21864]: 49d.17:40:41.436 [C] Platform------: RecoverFromRcpFailure() at radio_spinel_impl.hpp:2259: Failure
```
No mention of TCP errors now but still "too many RCP failures".

canisLupus1313, 2022-04-11:
@marcorussi I asked around internally about the revision of board that You are using. unfortunatelly those aren't available any whare in our offices since 0.9.0 was a preview revision. With my 2.0.1 this issue do not reproduce. Is there any chance that You can try this on newer version?

marcorussi, 2022-04-14:
@canisLupus1313 sorry for the late reply but I had few days off and meanwhile I got a new nRF52840 USB dongle. With this dingle I can run RCP reliably enough to say that the my problem had to do with the nRF52840 dev kit and probably with that specific revision. I came across another issues with the dongle though so I'll eventually raise another ticket about it. Would you like me to close this open ticket? I don't need to fix this particular issues anymore but I can investigate further in future if you like. Thanks.

canisLupus1313, 2022-04-14:
@marcorussi  thanks for confirmation that this is HW issue. I think we can close this issue then.

closed 2022-04-14 by canisLupus1313