openthread/ot-nrf528xx #322: NRF52840 dongle RCP via UART

fcgdam, 2022-05-15:
I've successfully compiled and used RCP through USB on the NRF52840 PCA10059 dongle to connect to the border router. 

Now I'm testing the same hardware but instead using an UART connection by selecting **UART_trans** instead of **USB_trans**:
`script/build nrf52840 UART_trans -DOT_BOOTLOADER=USB -DOT_THREAD_VERSION=1.2
`
While it compiles and flashes fine I have no success in making it working. I've changed the transport_config.h file to disable hardware flow control and also defined my RX and TX pins, but using a serial converter to USB for testing the UART communications yields no response from the RCP.

`
#ifndef UART_HWFC_ENABLED
#define UART_HWFC_ENABLED 0
#endif

#ifndef UART_PIN_TX
#define UART_PIN_TX 31
#endif

#ifndef UART_PIN_RX
#define UART_PIN_RX 29
#endif

`

Using **pyspinel** with the dongle when flashed with the USB variant works (at least the version command: _OPENTHREAD/thread-reference-20200818-1524-g7ec658e47; NRF52840; Feb 16 2022 11:57:20_), but not with the combination UART+converter. No answer whatsoever. 

The openthread commit used is **7ec658e47a39e1c3c55651e584e0748b259d2d4d**

Any ideas? Thanks for the help!

jwhui, 2022-05-16:
I have tested UART transport using an [nRF52840-DK](https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dk) and everything works as expected. If you have access to an nRF52840-DK, you could try that as well.

I would also suggest posting on [Nordic DevZone](https://devzone.nordicsemi.com/) to see if there are any other ideas.

Maybe @edmont has some ideas as well.

fcgdam, 2022-05-17:
I've build the RCP from the Openthread coprocessor sample from the Zephyr main repo and from the Nordic Connect SDK (it is the same, but Zephyr version is different), and it works fine using the UART. But if this repo works with the DK it might be something related with mine configuration (UART pins?).

closed 2022-05-17 by fcgdam