openthread/ot-nrf528xx #343: Problem with attaching devices

BAzili073, 2022-06-08:


jwhui, 2022-06-08:
From the logs, it appears that there are many transmission failures with no-ack errors. This could be due to poor RF connectivity between the devices.

Which nRF52840 devices are you using? What is the received signal strength being reported for received messages in the OpenThread logs on the RPi?

BAzili073, 2022-06-09:
Thanks for such fast answer!
We use devices of our own production based on nRF52840
Here is the log from RPi.
[RPi.log](https://github.com/openthread/ot-nrf528xx/files/8866909/RPi.log)
 | PAN  | MAC Address      | Ch | dBm | LQI |
+------+------------------+----+-----+-----+
| f0c6 | 02925165732a45d4 | 11 | -50 | 172 |
| f0c6 | a6678815ccf4d899 | 11 | -46 | 184 |
| f0c6 | e6fd73c544781877 | 11 | -60 | 124 |
| f0c6 | bafd8b412e0d32cd | 11 | -55 | 148 |

jwhui, 2022-06-13:
@BAzili073 , do you have any nRF52840-DK or USB dongles to test with? It would be interesting to see if you can replicate the issue with Nordic dev boards.

BAzili073, 2022-06-15:
Yeah, we will try to reproduce it on nRF52840-DK

jwhui, 2023-06-28:
Closing stale issue.

closed 2023-06-28 by jwhui