openthread/ot-nrf528xx #397: Adafruit Feather nRF52840 not shown as USB Device

fklement, 2022-09-02:
After I have successfully flashed the hex file to the device, it no longer appears on my Mac.

Flashing works fine:
```
>> nrfjprog -f nrf52 --chiperase --program ot-cli-ftd.hex --reset --verify                                                                                     ✔
Parsing image file.
Verifying programming.
Verified OK.
Applying system reset.
Run.
```

Before flashing open thread the feather device shows up under my `/dev/tty*` devices.
After that, unfortunately, no more. Does anyone have any idea what the problem is or what I am doing wrong?

canisLupus1313, 2022-09-22:
@fklement the board that You are compiling for is not supported in this repository. (Building with `-DOT_BOOTLOADER=USB` IHMO won't help.)
Have You tried using NCS repo instead?
https://github.com/nrfconnect/sdk-nrf
I can see that zephyr should support this board and so NCS.

jwhui, 2023-02-06:
Closing stale issue.

closed 2023-02-06 by jwhui