openthread/ot-nrf528xx #424: [mdk] Request to update mdk to an up-to-date version

Irving-cl, 2022-10-11:
## Background
The latest nRF52833 silicon revision uses the new hardened approtect mechanism (see [IN149 Informational Notice v1.1](https://infocenter.nordicsemi.com/pdf/in_149_v1.1.pdf?cp=4_1_2_2).) With this new revision it is no longer enough to have approtect disabled in UICR, we also need to unlock it as a part of your startup code. This unlocking is usually taken care of by the startup file. That requires us to use an up to date version of the MDK.

The specific issue we met: If we built a Thread FW with the current MDK for nrf52833, and we connect the debugger, APProtect will be triggered. I think a reasonable solution is to update the mdk version. Not sure which exact version solves this issue. Any version that does the unlock would be good.

Or if there are any other easier solutions for this issue, I would love to accept them.

edmont, 2022-10-13:
@Irving-cl could you give a try to this branch? https://github.com/openthread/ot-nrf528xx/pull/429

According to IN-149 rev 1.1:

> 2. Program code compiled with MDK 8.40.2 or later, with ENABLE_APPROTECT defined.

Bumping nrfx to 1.9.0 brings MDK 8.40.3, so this should fix your issues without major impact.

Irving-cl, 2022-10-26:
I've verified the PR. Thanks!

closed 2022-10-26 by Irving-cl