openthread/ot-nrf528xx #437: E104-BT5040u USB Stick does not enter DFU mode after initial flashing

VollMich, 2022-10-21:
Hi everybody, 

thank you for your great work. I did buy a E104-BT5040u since it comes with a plastic shell and it was advertised that the function and pinout of the module corresponds to the official Nordic nRF52840 USB Dongle module. The device has a blue shell that can be removed. There are two hardware buttons. I have put some photos of the dongle [here](https://imgur.com/a/c9voOdL). In factory state it booted right into DFU mode, without any buttons, as advertised by the supplier. the led was slowly blinking. I was able to build the firmware with the following options: 

`script/build nrf52840 USB_trans -DOT_BOOTLOADER=USB -DOT_THREAD_VERSION=1.2`

basically i followed [this slightly outdated guide.](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.7.0/matter/openthread_rcp_nrf_dongle.html) I was able to write the firmware to the dongle with:

` nrfutil dfu usb-serial -pkg build/bin/ot-rcp.zip -p /dev/ttyACM0`

It works seemingly flawlessly with otbr on my raspberry Pi. But now i want to build the firmware with different options, since i made a mistake and forced Thread version 1.2. following that outdated guide. I would rather use version 1.3 or omit this parameter completely. I have found this configuration [here](https://github.com/openthread/ot-nrf528xx/issues/257).

`./script/build nrf52840 USB_trans -DOT_BORDER_AGENT=ON -DOT_BORDER_ROUTER=ON -DOT_COMMISSIONER=ON -DOT_UDP_FORWARD=ON -DOT_LINK_RAW=ON`

the new image could be built without problems. 

Now my problem: 

I can't get into DFU mode anymore. No matter what buttons i push or push and hold while connecting, the device does not start DFU Mode. There is no blinking LED coming up. In Windows the stick always shows up as Open Thread Device no matter what i do. In factory state with factory DFU-only Software it showed up as usb serial device. It still works as RCP device in otbr on my Pi (paramter --radio-url spinel+hdlc+uart:///dev/ttyACM0). But i cant put the device into DFU mode anymore. Also: i cant connect to /dev/tttyACM0 with nrfutil in this state.

Am i missing something? is it possible that i disabled DFU bootloader capability in my stick somehow? I didn't really understand what i was doing since i was sure that the bootloader would always come up. But it doesn't. Is there anything that i can do to get into DFU mode again? Or maybe the device software contains DFU capability but on a different pin? The stick is advertised [here](https://www.ebyte.com/en/product-view-news.html?id=1185). there is also a manual [here](https://www.ebyte.com/en/downpdf.aspx?id=1185).

Or could i use some kind of SPI via usb to switch the device into DFU mode? I tried to connect to the device with NRFConnect with no luck. The LED does not start blinking. Also i have found [this](https://devzone.nordicsemi.com/f/nordic-q-a/75142/nrf52840-dongle-usb-dfu-problem) but there is no solution. It is mentioned that _The way the board enter DFU mode is by checking if the booting was from a pin reset or not. The pin reset is controlled by either the button SW2 or by the GPIO pin P0.19 (and also P0.21, P0.23, P0.25). Please avoid using these pins in your application._  but please don't follow me down into this rabbit hole. It's not my exact hardware anyways! 

Nordic speaks of enabling custom firmware with some kind of DFU support. where there it isn't needed to push a button at all. Did anybody concider this? Or is this firmware enabled to support DFU in a different way?

If there is no simple, obvious solution, i will obtain a different stick. I would love to have a stick with a plastic shell. I wouldn't want a whole clumsy dev kit. I want something sleak to attach to my raspberry pi. Any recommendations welcome. 

Sorry for this long post. I hope there is a short answer to my problem. Thank you very much!

Faithfully, 
Stefan

VollMich, 2022-10-21:
I guess I should have used the option

`-DOT_BOOTLOADER=UART`

Instead.  The firmware now has 

`DOT_BOOTLOADER=USB`

Enabled. The mystery is, that i can't even see the device with nrf connect in the windows SDK and it's programmer. The Driver is installed, it installed right automatically when i plugged it in for the first time. Before flashing with nrfutil on a different platform.

Now for dummies: 

`-DOT_BOOTLOADER=UART`

Would have worked wouldn't it? And i could probably have used nrfutil over and over again?

VollMich, 2022-10-21:
[This ](https://imgur.com/28ve0zR )is what happens when i try to use the NRF Programmer.The Windows Device Manager [shows ](https://imgur.com/a/OCAtrSl)COM6 and the Open Thread Device with an exclamation mark. I would switch to `-DOT_BOOTLOADER=UART` and use nrfutil from now on.  But i'm stuck with NRF in windows not communicating with the stick properly? I swear the fimware on the stick is built with `script/build nrf52840 USB_trans -DOT_BOOTLOADER=USB -DOT_THREAD_VERSION=1.2`

Any recommendations to get it working with NRF Programmer? Am i missing a driver? I'm running windows 10 and it was initlally stated that a driver had been installed for my device. But i think it was the driver for the serial mode of the initial DFU firmware that came with the stick.

VollMich, 2022-10-22:
I believe that the bootloader on my E104-BT5040u was overridden because i follwed [the outdated guide from nordic ](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.7.0/matter/openthread_rcp_nrf_dongle.html) and generated a  package using this command:

```
$ nrfutil pkg generate --hw-version 52 --sd-req=0x00 \
    --application build/bin/ot-rcp.hex \
    --application-version 1 build/bin/ot-rcp.zip
```
this command does not include the option `"--bootloader"`, what other sources occasionally do. This looks suspicious to me. I guess the firmware was flashed to my stick without the neccesary offset 0x1000 and thereby destroyed the DFU mode on my stick. 

I will consider the following steps: 

- buy a new E104-BT5040u
- onboard the stick with the nRFConnect utility (to be sure to have the latest and official Bootloder)
- flash the Firmware using nRFConnect checking for 0x1000 offset and never use nrfutil again 
- or somehow have a better understanding of nrfutil

In the [project readme](https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md) is this note:
_For new projects, you should use [nRF Connect SDK](https://www.nordicsemi.com/Products/Development-software/nRF-Connect-SDK). nRF5 SDK for Thread and Zigbee is a legacy solution and only provided in maintenance mode._ 
Question: Am I supposed to build this project, but with the nRF-Connet-SDK? Or am i supposed to use their coprocessor?

I am considering to flash the official [RCP Firmware example](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/openthread/coprocessor/README.html) from nordic. or [Zephyr](https://docs.zephyrproject.org/latest/samples/net/openthread/coprocessor/README.html).

Question: Will openthread be compatible with a stick that contains the RCP Firmware example or the coprocessor from Zephyr? It's also menbetter suited because of crypto-Acceleration support and bells and whistles. But how about compatibility as RCP for the latest release of openthread? I think it's not really clear which way to go. Is the RCP approach still a work-in-progress? Is the RCP approach in this project different to the RCP Approach from nordics example?

jwhui, 2023-02-08:
I would suggest reaching out via [Nordic DevZone](https://devzone.nordicsemi.com/).

closed 2023-02-08 by jwhui