openthread/ot-nrf528xx #549: Missing content in CLI output of an OpenThread FTD running on nRF52840 DK

superwhd, 2023-04-18:
1. I built an FTD for nRF52840 DK: 
```
./script/build nrf52840 UART_trans -DOT_DNS_CLIENT=ON -DOT_SRP_CLIENT=ON -DOT_ECDSA=ON  -DOT_CHANNEL_MONITOR=ON -DOT_BORDER_ROUTER=ON
```
Commit ID: 7aef51528d6ed996b89cdb004552eb84d7a2292c
2. By CLI, I let the board join a Thread network and ran `netdata show` for a few rounds, then I saw characters missing from the output **sometimes**. 
```
> netdata show
Prefixes:
fdfa:1e4c:713a:1::/64 paos low cc00
Routes:
fd31:d9cf:2e2f:fdf8::/64 s med cc00
<--------------------------------------------------- Missing 'Services'
44970 5d fd6b88856fef000056581ba57516edf3d11f s cc00
44970 01 10000500000e10 s cc00
Done
```
```
> netdata show
Prefixes:
fdfa:1e4c:713a:1::/64 paos low cc00
Routes:
 s med cc00                             <---------------------------- Missing the prefix at the beginning.
Services:
44970 5d fd6b88856fef000056581ba57516edf3d11f s cc00
44970 01 10000500000e10 s cc00
Done
```
Full log: [screenlog.log](https://github.com/openthread/ot-nrf528xx/files/11259531/screenlog.log)

3. On the contrary, I tried the `netdata show` on my nRF52840 dongle as well and didn't see the issue. I'm not sure what commit version of the dongle was using though.

jwhui, 2023-04-18:
@superwhd , did you disable Mass Storage Device (MSD)?

https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md#mass-storage-device-known-issue

superwhd, 2023-04-19:
@jwhui I tried disabling the MSD but the issue is still happening.

I ran the following and power cycled the board.
```
JLinkExe -device NRF52840_XXAA -if SWD -speed 4000 -autoconnect 1 

MSDDisable
```

jwhui, 2023-04-19:
Maybe @LuDuda @edmont have some ideas?

LuDuda, 2023-04-19:
@superwhd Just to confirm.. did you power cycle the DK after `MSDDisable` command? Also what is your revision of J-Link on Host and DK?

superwhd, 2023-04-19:
@LuDuda 

I've power cycled the board after `MSDDisable` and the issue was still there. Is there a way to verify `MSD` is disabled?

Here's the initial output of `JLinkExe`:
```
JLinkExe -device NRF52840_XXAA -if SWD -speed 4000 -autoconnect 1
SEGGER J-Link Commander V7.54c (Compiled Sep 20 2021 13:59:37)
DLL version V7.54c, compiled Sep 20 2021 13:59:21

Connecting to J-Link via USB...O.K.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jan 20 2022 16:43:31
Hardware version: V1.00
S/N: 683316439
License(s): RDI, FlashBP, FlashDL, JFlash, GDB
VTref=3.300V
Device "NRF52840_XXAA" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
Found SW-DP with ID 0x2BA01477
DPIDR: 0x2BA01477
Scanning AP map to find all available APs
AP[2]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
AP[1]: JTAG-AP (IDR: 0x02880000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
[0][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
[0][1]: E0001000 CID B105E00D PID 003BB002 DWT
[0][2]: E0002000 CID B105E00D PID 002BB003 FPB
[0][3]: E0000000 CID B105E00D PID 003BB001 ITM
[0][4]: E0040000 CID B105900D PID 000BB9A1 TPIU
[0][5]: E0041000 CID B105900D PID 000BB925 ETM
Cortex-M4 identified.
```

edmont, 2023-04-19:
Thanks @superwhd, I was able to reproduce it on my side with a single device:

```
> panid 0xface
Done
> networkkey 00112233445566778899aabbccddeeff
Done
> ifconfig up
Done                                                                                                                                                                                                                                                               
> thread start                                                                                                                                                                                                                                                     
Done     
> prefix add 2002::/64 pro med                                                                                                                                                                                                                                     
Done    
> dataset commit active                                                                                                                                                                                                                                            
Done   
> netdata show                                                                                                                                                                                                                                                     
Prefixes:                                                                                                                                                                                                                                                          
2002:0:0:0::/64 pro med ec00                                                                                                                                                                                                                                       
Routes:                                                                                                                                                                                                                                                            
Services:  
Done
[...]
> netdata show                                                                                                                                                                                                                                                     
Prefixes:                                                                                                                                                                                                                                                          
 pro med ec00                     <---------------------------- Missing the prefix at the beginning.                                                                                                                                                                                                         
Routes:                                                                                                                                                                                                                                                            
Services:                                                                                                                                                                                                                                                          
Done 
```

Are we sure this is a Nordic-only issue?

canisLupus1313, 2023-04-20:
@superwhd Which version of dk are You using?

superwhd, 2023-04-20:
> @superwhd Which version of dk are You using?

PCA10056
1.0.0 
2018.19

canisLupus1313, 2023-04-20:
@superwhd  please try to reproduce this issue with `USB_trans` ?
Also please refer to https://github.com/openthread/openthread/issues/7590
I will try to check which versions of dk were affected.

canisLupus1313, 2023-04-20:
@superwhd The working version of DK should be 3.0.0 (older versions have Jlink based on slow MCU which causes problems that You observe). You can try to update the Jlink version but this won't fix the issue but it might improve the situation a little.

superwhd, 2023-04-27:
@canisLupus1313 I got a 3.0.0 DK and updated my JLink to V7.86h. The issue seems still existing. 
[DK3.log](https://github.com/openthread/ot-nrf528xx/files/11341365/DK3.log)
By searching the text in the log, I found 92 `netdata show` calls while there are 84 correct responses.

Output of JLink:
```
SEGGER J-Link Commander V7.86h (Compiled Apr 12 2023 16:10:01)
DLL version V7.86h, compiled Apr 12 2023 16:09:43

Connecting to J-Link via USB...O.K.
Firmware: J-Link OB-nRF5340-NordicSemi compiled Nov  7 2022 16:22:01
Hardware version: V1.00
J-Link uptime (since boot): 0d 00h 06m 06s
S/N: 1050271794
License(s): RDI, FlashBP, FlashDL, JFlash, GDB
USB speed mode: Full speed (12 MBit/s)
VTref=3.300V
Device "NRF52840_XXAA" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end - Took 1.78ms
Found SW-DP with ID 0x2BA01477
DPIDR: 0x2BA01477
CoreSight SoC-400 or earlier
Scanning AP map to find all available APs
AP[2]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
AP[1]: JTAG-AP (IDR: 0x02880000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
[0][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
[0][1]: E0001000 CID B105E00D PID 003BB002 DWT
[0][2]: E0002000 CID B105E00D PID 002BB003 FPB
[0][3]: E0000000 CID B105E00D PID 003BB001 ITM
[0][4]: E0040000 CID B105900D PID 000BB9A1 TPIU
[0][5]: E0041000 CID B105900D PID 000BB925 ETM
Memory zones:
  Zone: "Default" Description: Default access mode
Cortex-M4 identified.
```

canisLupus1313, 2023-04-27:
@superwhd This seems odd I will take a look into it.

canisLupus1313, 2023-04-28:
@superwhd Please try with this #556

superwhd, 2023-05-04:
Thanks, @canisLupus1313! With this fix I tried `netdata show` 90 times on a 1.0.0 DK board in UART mode and nothing missed in the output.

closed 2023-05-04 by jwhui