openthread/ot-nrf528xx #623: OT_ERROR_NOT_FOUND when canonical names received from otDnsClientResolveIp4Address()

tom-brown-kr, 2023-08-03:
I am using an nRF52840 SoC and a Raspberry Pi border router with nat64, dns, and snp enabled. When I use otDnsClientResolveIp4Address() to get the ipv4 address of an NTP sever (ex. "us.pool.ntp.org"), I receive only type A responses and otDnsAddressResponseGetAddress() is able to get any of the four returned addresses. I am able to do an NTP update successfully from any of those.
[image]

When I attempt to resolve an Azure microservice hostname, I receive two CNAME entries followed by the desired ipv4 entry. However, otDnsAddressResponseGetAddress() in my otDnsAddressCallback reports OT_ERROR_NOT_FOUND for all entries in the list of addresses.
[image]
```
static void otHTTPDNScallback(otError aError, const otDnsAddressResponse *aResponse, void *aContext)
{
	otError oterr = OT_ERROR_NONE;
	char tmpbuf[50];
	int i=0;

	if (aError==OT_ERROR_NONE)
	{
		while (oterr!=OT_ERROR_NOT_FOUND)
		{
                        //aContext is a pointer to a buffer to store the returned ipv6 address (border router's ipv6 prefix + DNS ipv4 address)
			oterr = otDnsAddressResponseGetAddress(aResponse, i, aContext, NULL);
			
			if (oterr==OT_ERROR_NONE)
			{
				otIp6AddressToString(aContext, tmpbuf, sizeof(tmpbuf));
				LOG_INF("DNS returned IP addr #%d: %s",i,tmpbuf);
                                break; //use first found IP address
			}
                        else
                        {
                                LOG_ERR("Error in otDnsAddressResponseGetAddress (err %d)", oterr);
                        }
			i+=1;
	         }
	}
        else
        {
                 LOG_ERR("Error in HTTP DNS lookup (err: %d)", aError);
        }
	k_sem_give(&getHTTPaddr_sem); //Need to hold main function until address is obtained in this callback. Return semaphore here so program can continue
}
```

However, if I do a DNS request for the last CNAME - "my-service.azuremicroservices.io" - then I do not receive any CNAME results and otDnsAddressResponseGetAddress() is able to successfully get the IP address 12.34.56.78 (OT_ERROR_NONE). I can then connect successfully to that IP address. The problem here is that I don't initially know what this last CNAME is.

Am I doing something incorrectly or is this a bug? How can I get the IP address in the last entry?

abtink, 2023-08-03:
@tom-brown-kr I have a guess/theory of why it is not working.

The CNAME RR are used to provide a different name for a host name.
In the response there seems to two levels CNAME name changes happening:
- The name you queried is mapped to `-support.private-...` and then the `-support.private...` name is again mapped to `.azuremicro...` name and A record is provided for this name

In  the `Dns::Clinet` implementation we do check for one level of indirection (we check if there is `CNAME` and we use the new name instead  in [`Client::Response::CheckForHostNameAlias`](https://github.com/openthread/openthread/blob/499f6680a3afd600fdd8f2d003ea533334b4741f/src/core/net/dns_client.cpp#L167)) but not if there are multiple CNAME records and multiple name changes. We can add this in the code to support this.

tom-brown-kr, 2023-08-03:
Thank you for the quick investigation - support for multiple CNAMEs would be greatly appreciated. I will be sure to read through `dns_client.cpp` in more detail for my own education.

Do you know approximately how long it might take to commit an update?

tom-brown-kr, 2023-08-03:
@abtink As it turns out, `CheckForHostNameAlias()` is written in a way that is very easy to iterate! The `// <-` comments below indicate my changes and additions.

```
Error Client::Response::CheckForHostNameAlias(Section aSection, Name &aHostName) const
{
    // If the response includes a CNAME record mapping the query host
    // name to a canonical name, we update `aHostName` to the new alias
    // name. Otherwise `aHostName` remains as before.

    Error       findRecError, error; // <- added findRecError
    uint16_t    offset;
    uint16_t    numRecords;
    CnameRecord cnameRecord;
    int i=0;
    const int i_maxcounter = 20;

    VerifyOrExit(mMessage != nullptr, error = kErrorNotFound);

    do { // <- added do-while loop to iterate through one or more CNAME record(s). i_maxcounter prevents infinite while loop
        SelectSection(aSection, offset, numRecords);
        error = ResourceRecord::FindRecord(*mMessage, offset, numRecords, /* aIndex */ 0, aHostName, cnameRecord);

        findRecError = error; // <- use findRecError to avoid interfering with current returning of "error"

        switch (error)
        {
        case kErrorNone:
            // A CNAME record was found. `offset` now points to after the
            // last read byte within the `mMessage` into the `cnameRecord`
            // (which is the start of the new canonical name).
            aHostName.SetFromMessage(*mMessage, offset);
            error = Name::ParseName(*mMessage, offset);
            break;

        case kErrorNotFound:
            error = kErrorNone;
            break;

        default:
            break;
        }

        i+=1;
    } while(findRecError==kErrorNone && i<=i_maxcounter);   
exit:
    return error;
}
```

If I modify the comments and double-check your guidelines/conventions, could I submit a pull request for this? Is there anything you would recommend changing?

abtink, 2023-08-03:
Just saw this. I am actually adding support for this in the code now. Will submit a PR soon.

abtink, 2023-08-03:
Submitted the PR below which should help address this:
- https://github.com/openthread/openthread/pull/9339

@tom-brown-kr it would be great if you can cherry-pick this and test it with your setup, so we can validate it :)

tom-brown-kr, 2023-08-03:
@abtink I have confirmed that your PR #9339 solves my issue. Thank you again for the quick assistance!

closed 2023-08-03 by tom-brown-kr

abtink, 2023-08-03:
@tom-brown-kr, awesome. Thanks  helping validate the PR quickly. :)