openthread/ot-nrf528xx #668: /opt/nrfjprog/: No such file or directory

tunm1228, 2023-10-13:
[image]

superwhd, 2023-10-13:
I wonder if you've done [this step](https://openthread.io/codelabs/openthread-hardware#install-nrf5x-command-line-tools) to install the NRF5x command line tools? `nrfjprog` is part of the tools.

tunm1228, 2023-10-13:
> I wonder if you've done [this step](https://openthread.io/codelabs/openthread-hardware#install-nrf5x-command-line-tools) to install the NRF5x command line tools? `nrfjprog` is part of the tools.

I haven't done this step yet. I will try. thank you

tunm1228, 2023-10-13:
> I wonder if you've done [this step](https://openthread.io/codelabs/openthread-hardware#install-nrf5x-command-line-tools) to install the NRF5x command line tools? `nrfjprog` is part of the tools.

May I ask you. Download and push in and do what is in this downloaded file?

tunm1228, 2023-10-13:
I installed it but still can't find this folder
[image]

tunm1228, 2023-10-13:
[image]

superwhd, 2023-10-13:
Could you check `which nrfjprog`? Probably the executable binary already in your `$PATH`.

tunm1228, 2023-10-13:
[image]
it is here

superwhd, 2023-10-13:
Then you should be able to run `nrfjprog`. Is there specifically anything else you need in the `nrfjprog/` directory?

tunm1228, 2023-10-13:
Project ot-nrf528xx is in the root directory. Can you guide me to navigate to nrfjprog

tunm1228, 2023-10-13:
[image]
it had this error

superwhd, 2023-10-13:
Try removing `./` from the beginning of your command.

tunm1228, 2023-10-13:
[image]
still error

tunm1228, 2023-10-13:
[image]

tunm1228, 2023-10-13:
[image]

tunm1228, 2023-10-16:
> Try removing `./` from the beginning of your command.

It looks like read back protection (APPROTECT) is enabled. If so, you can disable it using "nrfjprog --recover".
[image]

closed 2023-10-16 by superwhd

tunm1228, 2023-10-16:
@superwhd Do you know what this error is caused by?
[image]

tunm1228, 2023-10-16:
I had to unplug and plug it back in for it to work
[image]