openthread/ot-nrf528xx #926: Bootstrap script fails

Slyke, 2024-11-01:
Following the instructions from: https://openthread.io/guides/build/index.md

```
$ docker run -it --rm openthread/environment bash
root@d8807b5908b6:/# pwd
/
root@d8807b5908b6:/# ls
bin  boot  dev  etc  home  lib  lib32  lib64  libx32  media  mnt  openthread  opt  proc  root  run  sbin  srv  sys  tmp  usr  var
root@d8807b5908b6:/# ls openthread/
AUTHORS  build  BUILD.gn  CMakeLists.txt  CODE_OF_CONDUCT.md  CONTRIBUTING.md  doc  etc  examples  include  LICENSE  NOTICE  README.md  script  SECURITY.md  src  STYLE_GUIDE.md  tests  third_party  tools  zephyr
root@d8807b5908b6:/# git clone https://github.com/openthread/ot-nrf528xx.git
Cloning into 'ot-nrf528xx'...
remote: Enumerating objects: 3308, done.
remote: Counting objects: 100% (797/797), done.
remote: Compressing objects: 100% (187/187), done.
remote: Total 3308 (delta 677), reused 714 (delta 610), pack-reused 2511 (from 1)
Receiving objects: 100% (3308/3308), 5.58 MiB | 10.00 MiB/s, done.
Resolving deltas: 100% (1815/1815), done.
root@d8807b5908b6:/# cd ot-nrf528xx/
root@d8807b5908b6:/ot-nrf528xx# ./script/bootstrap
++ dirname ./script/bootstrap
+ ./script/../openthread/script/bootstrap
./script/bootstrap: line 32: ./script/../openthread/script/bootstrap: No such file or directory
root@d8807b5908b6:/ot-nrf528xx# ls
CMakeLists.txt  LICENSE  openthread  README.md  script  src  third_party
root@d8807b5908b6:/ot-nrf528xx# ls openthread/
root@d8807b5908b6:/ot-nrf528xx# ls script/
bootstrap  build  make-pretty  test
```

Slyke, 2024-11-01:
Workaround:

After cloning and cd'ing into `ot-nrf528xx`, this seems to work:
```
$ rm -rf openthread/
$ cp -r /openthread/ .
```

Then run:
```
./script/bootstrap
```

jwhui, 2024-11-01:
Did you run the following command before bootstrap?
```
$ git submodule update --init
```
See https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md#prerequisites

Slyke, 2024-11-02:
Thank also works!

closed 2024-11-02 by Slyke