openthread/ot-nrf528xx #1023: Starting With IfConfig Down And Thread Stoped

ImAgremlin1123, 2025-04-09:
Hi, I'm trying to use the nRF52840 as a repeater for my thread network. The goal is to be able to plug the nRF into a wall outlet and have it act as a router. However Ive noticed that the thread network always starts with its IfConfig Down and Thread stopped so it wont automatically connect to a network without manually starting it through CLI. Ive tried CLI-Radio, CLI-FTD, and NCP-FTD.

Is there a file somewhere i can change how it handles boot ups, or is there a newer more updated repository with assess to CLI commands like auto start

nasserma, 2025-05-19:
I have a solution for this, but it is not very elegant, based on adding some code to the CLI application found in the openthread code (ot-nrf528xx/openthread/examples/apps/cli/main.c). Essentially you can add code using the top-level functions of the CLI server by declaring a null terminated string and then emulate the user connecting to the dongle and interfacing with the CLI using a terminal emulator,

`char ifconfig_string[255] = "ifconfig up";

otCliInputLine(ifconfig_string);`

for this to work you need to execute the "tasklet" and driver processes between every input string to the CLI server,

`otTaskletsProcess(instance);
otSysProcessDrivers(instance);`

I have confirmed this works but have not written the code in a very clean way, I'll post that code once I have tested it more and included some sort of watchdog.

jwhui, 2025-05-21:
You could also call the APIs directly:

```diff
diff --git a/examples/apps/cli/main.c b/examples/apps/cli/main.c
index 2ede597cd..6b1de33a8 100644
--- a/examples/apps/cli/main.c
+++ b/examples/apps/cli/main.c
@@ -36,8 +36,11 @@
 #include <openthread/config.h>
 
 #include <openthread/cli.h>
+#include <openthread/dataset.h>
 #include <openthread/diag.h>
+#include <openthread/ip6.h>
 #include <openthread/tasklet.h>
+#include <openthread/thread.h>
 #include <openthread/platform/logging.h>
 #include <openthread/platform/misc.h>
 
@@ -148,6 +151,14 @@ pseudo_reset:
     IgnoreError(otPlatLogCrashDump());
 #endif
 
+#if OPENTHREAD_FTD || OPENTHREAD_MTD
+    if (otDatasetIsCommissioned(instance))
+    {
+        IgnoreError(otIp6SetEnabled(instance, true));
+        IgnoreError(otThreadSetEnabled(instance, true));
+    }
+#endif
+
     while (!otSysPseudoResetWasRequested())
     {
         otTaskletsProcess(instance);
```